prusa3d/PrusaSlicer #2617: [Feature Request] Remove wipe tower

jwillmer, 2019-07-11:
Why does the wipe tower need to be the same height as the object? Can't we just move the print head down to the print bed if a color change is in progress? Or give the user an option to manually make sure the right color is coming out now - as you do in the normal filament loading process.

lukasmatena, 2019-07-11:
1. If the wipe tower was lower, travels to it would add a lot to the print time (z movement is quite slow). Maybe more importantly, it would also open the possibility of collision with the object. Not that it is impossible to solve but I'm not sure how many people would actually use it.

2. Manual tool changes should be possible soon, at least to some extent. We are currently testing an improved version of MM handling that would allow you to disable the wipe tower and use custom gcode for the tool change. That might cover it (for MM printers at least).

jwillmer, 2019-07-11:
1) Yes it would slow down the time for changing the filament but I think it would still be faster then printing the whole wipe tower.
2) Cool, happy to hear.

AmyTheCute, 2019-07-12:
I think the smarter solution is just to make a purge bucket. I'm sure the prusa company is already thinking of that!

there are tutorials online for it already so you can search around.

josefprusa, 2019-07-17:
Waste buckets are terrible. Waste tower makes the excess filament compact, but the same amount of material in bucket can have upto 10x the volume. You need to clean it multiple times during the printing. We started with waste bucket btw.

AmyTheCute, 2019-07-17:
> Waste buckets are terrible. Waste tower makes the excess filament compact, but the same amount of material in bucket can have upto 10x the volume. You need to clean it multiple times during the printing. We started with waste bucket btw.

Huh. That's cool to know :)

I did see that one person on reddit/r/3dprinting that made a system that basically turned the waste into a single blob that just dropped into a bucket. it made them really compact! I'll keep on searching and see if I can find it to post it here :)

AmyTheCute, 2019-07-17:
> Waste buckets are terrible. Waste tower makes the excess filament compact, but the same amount of material in bucket can have upto 10x the volume. You need to clean it multiple times during the printing. We started with waste bucket btw.

Hey I know you're probably know far more than me but just in case I found the link:
https://www.reddit.com/r/3Dprinting/comments/c4rfky/raving_rabbids_timelapse_multicolor_painted_and/

this turns the waste into little compact balls instead of a hairy mess!

lukasmatena, 2019-12-17:
The "low wipe tower" part of this request was implemented for PrusaSlicer2.2.0 release. Our testing showed that my guess about huge impact on print time was unjustified - the actual print time is usually similar to what it was before, sometimes even shorter. It is left to the user to make sure there are no collisions with the print.

closed 2019-12-17 by jwillmer