prusa3d/PrusaSlicer #2618: Prusa Slicer cannot import config from gcode

jeffgros, 2019-07-11:
### Version
Version: PrusaSlicer-2.0.0+ win64-201905201652

### Operating system type + version
Operating system is Windows 10 64 bit

### 3D printer brand / version + firmware version (if known)
Prusa MK3S MMU2S (firmware not important, not a printer issue)

### Behavior
Prusa Slicer cannot import config from gcode

Steps to reproduce:
1. Download Prusa enclosure V2 3d print files: https://www.prusaprinters.org/prints/3673-mmu2s-enclosure/files
1. Open Prusa Slicer.
2. File->Import->Import Config.
3. Browse to directory with Prusa Enclosure V2 gcode.
4. Choose "prusa_mmu_enclosure_1_02mm_pet_mk3s_15h36m.gcode".
5. Error message pops up. "PrusaSlicer has encountered an error. Not a Slic3r generated g-code".

Note that I've had success with some gcode files that I created using the old Slic3rPE.

lukasmatena, 2019-07-11:
This seems to be a duplicate of #2304 and #2374, so it should be already fixed. The latter issue describes a possible workaround (thanks @mascalx) until the fix is released.

closed 2019-07-11 by lukasmatena

jeffgros, 2019-07-11:
Confirmed. Sorry for duplicate!