prusa3d/PrusaSlicer #2619: [Linux] Assertion failure after a failed firmware flash

mhelvens, 2019-07-11:
### Version
2.0.0

### Operating system type + version
Linux 4.19.56-1-MANJARO x86_64

### Behavior
* I did not have write permission to `/dev/ttyACM0`. (My fault, it was a new Linux install.)
* I tried to flash new firmware, got the appropriate error message.
* I clicked "Close".
* I got the assertion failure below.

```
ASSERT INFO:
./src/gtk/dialog.cpp(199): assert "Assert failure" failed in EndModal(): either wxDialog:EndModal called twice or ShowModal wasn't called

BACKTRACE:
[1] wxDialog::EndModal(int)
[2] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[3] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[4] wxEvtHandler::TryHereOnly(wxEvent&)
[5] wxEvtHandler::ProcessEventLocally(wxEvent&)
[6] wxEvtHandler::ProcessEvent(wxEvent&)
[7] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[8] wxWindowBase::Close(bool)
[9] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[10] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[11] wxEvtHandler::TryHereOnly(wxEvent&)
[12] wxEvtHandler::ProcessEventLocally(wxEvent&)
[13] wxEvtHandler::ProcessEvent(wxEvent&)
[14] wxDialogBase::EmulateButtonClickIfPresent(int)
[15] wxDialogBase::OnCloseWindow(wxCloseEvent&)
[16] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[17] wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*)
[18] wxEvtHandler::TryHereOnly(wxEvent&)
[19] wxEvtHandler::ProcessEventLocally(wxEvent&)
[20] wxEvtHandler::ProcessEvent(wxEvent&)
[21] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[22] wxWindowBase::Close(bool)
[23] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[24] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[25] wxEvtHandler::TryHereOnly(wxEvent&)
[26] wxEvtHandler::ProcessEventLocally(wxEvent&)
[27] wxEvtHandler::ProcessEvent(wxEvent&)
[28] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[29] g_closure_invoke
[30] g_signal_emit_valist
[31] g_signal_emit
[32] g_closure_invoke
[33] g_signal_emit_valist
[34] g_signal_emit
[35] g_closure_invoke
[36] g_signal_emit_valist
[37] g_signal_emit
[38] gtk_propagate_event
[39] gtk_main_do_event
[40] g_main_context_dispatch
[41] g_main_loop_run
[42] gtk_main
[43] wxGUIEventLoop::DoRun()
[44] wxEventLoopBase::Run()
[45] wxDialog::ShowModal()
[46] Slic3r::FirmwareDialog::run(wxWindow*)
[47] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[48] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[49] wxEvtHandler::TryHereOnly(wxEvent&)
[50] wxEvtHandler::ProcessEventLocally(wxEvent&)
[51] wxEvtHandler::ProcessEvent(wxEvent&)
[52] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[53] wxMenuBase::SendEvent(int, int)
[54] g_closure_invoke
[55] g_signal_emit_valist
[56] g_signal_emit
[57] gtk_widget_activate
[58] gtk_menu_shell_activate_item
[59] g_closure_invoke
[60] g_signal_emit_valist
[61] g_signal_emit
[62] gtk_propagate_event
[63] gtk_main_do_event
[64] g_main_context_dispatch
[65] g_main_loop_run
[66] gtk_main
[67] wxGUIEventLoop::DoRun()
[68] wxEventLoopBase::Run()
[69] wxAppConsoleBase::MainLoop()
[70] wxEntry(int&, wchar_t**)
[71] Slic3r::CLI::run(int, char**)
[72] main
[73] __libc_start_main
[74] _start
```

vojtechkral, 2019-07-12:
Thanks, will have a look...

mweinelt, 2019-09-07:
Also seen in v2.1.0rc after closing firmware flash dialog.

```
ASSERT INFO:
./src/gtk/dialog.cpp(199): assert "Assert failure" failed in EndModal(): either wxDialog:EndModal called twice or ShowModal wasn't called

BACKTRACE:
[1] wxDialog::EndModal(int)
[2] wxEventFunctorFunctor<wxEventTypeTag<wxCloseEvent>, Slic3r::FirmwareDialog::FirmwareDialog(wxWindow*)::{lambda(wxCloseEvent&)#11}>::operator()(wxEvtHandler*, wxEvent&)
[3] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[4] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[5] wxEvtHandler::TryHereOnly(wxEvent&)
[6] wxEvtHandler::ProcessEventLocally(wxEvent&)
[7] wxEvtHandler::ProcessEvent(wxEvent&)
[8] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[9] wxWindowBase::Close(bool)
[10] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[11] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[12] wxEvtHandler::TryHereOnly(wxEvent&)
[13] wxEvtHandler::ProcessEventLocally(wxEvent&)
[14] wxEvtHandler::ProcessEvent(wxEvent&)
[15] wxDialogBase::EmulateButtonClickIfPresent(int)
[16] wxDialogBase::OnCloseWindow(wxCloseEvent&)
[17] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[18] wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*)
[19] wxEvtHandler::TryHereOnly(wxEvent&)
[20] wxEvtHandler::ProcessEventLocally(wxEvent&)
[21] wxEvtHandler::ProcessEvent(wxEvent&)
[22] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[23] wxWindowBase::Close(bool)
[24] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[25] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[26] wxEvtHandler::TryHereOnly(wxEvent&)
[27] wxEvtHandler::ProcessEventLocally(wxEvent&)
[28] wxEvtHandler::ProcessEvent(wxEvent&)
[29] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[30] g_closure_invoke
[31] g_signal_emit_valist
[32] g_signal_emit
[33] g_closure_invoke
[34] g_signal_emit_valist
[35] g_signal_emit
[36] g_closure_invoke
[37] g_signal_emit_valist
[38] g_signal_emit
[39] gtk_propagate_event
[40] gtk_main_do_event
[41] g_main_context_dispatch
[42] g_main_loop_run
[43] gtk_main
[44] wxGUIEventLoop::DoRun()
[45] wxEventLoopBase::Run()
[46] wxDialog::ShowModal()
[47] Slic3r::FirmwareDialog::run(wxWindow*)
[48] Slic3r::GUI::GUI_App::add_config_menu(wxMenuBar*)::{lambda(wxEvent&)#4}::operator()(wxEvent&) const [clone .isra.638]
[49] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[50] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[51] wxEvtHandler::TryHereOnly(wxEvent&)
[52] wxEvtHandler::ProcessEventLocally(wxEvent&)
[53] wxEvtHandler::ProcessEvent(wxEvent&)
[54] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[55] wxMenuBase::SendEvent(int, int)
[56] g_closure_invoke
[57] g_signal_emit_valist
[58] g_signal_emit
[59] gtk_widget_activate
[60] gtk_menu_shell_activate_item
[61] g_closure_invoke
[62] g_signal_emit_valist
[63] g_signal_emit
[64] gtk_propagate_event
[65] gtk_main_do_event
[66] g_main_context_dispatch
[67] g_main_loop_run
[68] gtk_main
[69] wxGUIEventLoop::DoRun()
[70] wxEventLoopBase::Run()
[71] wxAppConsoleBase::MainLoop()
[72] wxEntry(int&, wchar_t**)
[73] Slic3r::CLI::run(int, char**)
[74] main
[75] __libc_start_main
[76] _start
```

vojtechkral, 2019-09-09:
@mhelvens @mweinelt We didn't manage to reproduce the problem, but I _think_ I know what the problem is (cf. the commit above).

Would you please re-test with the following build?
https://drive.google.com/drive/folders/1_VXGlBZmeyyPrV1EkLrnNqaUVG-7P7Sc?usp=sharing

bubnikv, 2019-09-11:
We hope the issue is fixed in PrusaSlicer 2.1.0-rc2. Please let us know whether the issue was resolved for you.

lukasmatena, 2019-11-29:
No response, closing. The issue should be fixed, no further bug reports regarding this were filed since.

closed 2019-11-29 by lukasmatena