prusa3d/PrusaSlicer #2620: Support z-contact distance errors causing poor overhangs.

AbeFM, 2019-07-11:
### Version
2.0.0
### Operating system type + version
win10

### 3D printer brand / version + firmware version (if known)
mk3smmu2s, 3.7.1

### Behavior
Supports leave an extra gap. I had this issue well over a year ago, read about #102 https://github.com/prusa3d/PrusaSlicer/issues/102, and a few others. Since it seemed well understood at the time, I assumed it would get fixed.

Here's some quick data collected:

z-contact-distance | Layer height | XY dist % of 0.2 | Support | Object | As printed Distance | Error | `

z-contact-distance | Layer height | XY dist % of 0.2 | Support | Object | As printed Distance | Error |  
-- | -- | -- | -- | -- | -- | -- | --
0 | 0.08 | 50 | 1.54 | 1.62 | 0 | 0 |  
0.001 | 0.08 | 50 | 1.22 | 1.62 | 0.32 | 0.319 |  
0.1 | 0.08 | 50 | 1.12 | 1.62 | 0.42 | 0.32 |  
0.2 | 0.08 | 50 | 1.02 | 1.62 | 0.52 | 0.32 |  
0.5 | 0.08 | 50 | 0.72 | 1.62 | 0.82 | 0.32 |  
  |   |   |   |   |   |   |  
z-contact-distance | Layer height | XY dist % of 0.2 | Support | Object | As printed Distance | Error |  
0 | 0.1 | 50 | 1.52 | 1.63 | 0.01 | 0.01 |  
0.001 | 0.1 | 50 | 1.22 | 1.63 | 0.31 | 0.309 |  
0.1 | 0.1 | 50 | 1.13 | 1.63 | 0.4 | 0.3 |  
0.2 | 0.1 | 50 | 1.02 | 1.63 | 0.51 | 0.31 |  
0.5 | 0.1 | 50 | 0.73 | 1.63 | 0.8 | 0.3 |  
z-contact-distance | Layer height | XY dist % of 0.2 | Support | Object | As printed Distance | Error |  
0 | 0.15 | 50 | 1.54 | 1.69 | 0 | 0 |  
0.001 | 0.15 | 50 | 1.24 | 1.69 | 0.3 | 0.299 |  
0.01 | 0.15 | 50 | 1.23 | 1.69 | 0.31 | 0.3 |  
0.05 | 0.15 | 50 | 1.19 | 1.69 | 0.35 | 0.3 |  
0.1 | 0.15 | 50 | 1.14 | 1.69 | 0.4 | 0.3 |  
0.2 | 0.15 | 50 | 1.04 | 1.69 | 0.5 | 0.3 |  
0.5 | 0.15 | 50 | 0.74 | 1.69 | 0.8 | 0.3 |  
  |   |   |   |   |   |   |  
z-contact-distance | Layer height | XY dist % of 0.2 | Support | Object | As printed Distance | Error |  
0 | 0.15 | 5 | 1.54 | 1.69 | 0 | 0 |  
0.05 | 0.15 | 5 | 1.19 | 1.69 | 0.35 | 0.3 |  
0.1 | 0.15 | 5 | 1.14 | 1.69 | 0.4 | 0.3 |  
0.2 | 0.15 | 5 | 1.04 | 1.69 | 0.5 | 0.3 |  
0.5 | 0.15 | 5 | 0.74 | 1.69 | 0.8 | 0.3 |  
  |   |   |   |   |   |   |  
z-contact-distance | Layer height | XY dist % of 0.2 | Support | Object | As printed Distance | Error |  
0 | 0.2 | 50 | 1.45 | 1.69 | 0.04 | 0.04 |  
0.001 | 0.2 | 50 | 1.25 | 1.69 | 0.24 | 0.239 |  
0.1 | 0.2 | 50 | 1.15 | 1.69 | 0.34 | 0.24 |  
0.2 | 0.2 | 50 | 1.05 | 1.69 | 0.44 | 0.24 |  
0.5 | 0.2 | 50 | 0.75 | 1.69 | 0.74 | 0.24 |  
  |   |   |   |   |   |   |  
z-contact-distance | Layer height | XY dist % of 0.2 | Support | Object | As printed Distance | Error | First Layer
0 | 0.3 | 50 | 1.58 | 1.88 | 0 | 0 | 125%
0.001 | 0.3 | 50 | 1.48 | 1.88 | 0.1 | 0.099 | 125%
0.1 | 0.3 | 50 | 1.38 | 1.88 | 0.2 | 0.1 | 125%
0.2 | 0.3 | 50 | 1.28 | 1.88 | 0.3 | 0.1 | 125%
0.5 | 0.3 | 50 | 0.99 | 1.88 | 0.59 | 0.09 | 125%
  |   |   |   |   |   |   |  
z-contact-distance | Layer height | XY dist % of 0.2 | Support | Object | As printed Distance | Error | First Layer
0 | 0.3 | 50 | 1.5 | 1.8 | 0 | 0 | 100%
0.001 | 0.3 | 50 | 1.41 | 1.8 | 0.09 | 0.089 | 100%
0.1 | 0.3 | 50 | 1.31 | 1.8 | 0.19 | 0.09 | 100%
0.2 | 0.3 | 50 | 1.21 | 1.8 | 0.29 | 0.09 | 100%
0.5 | 0.3 | 50 | 0.91 | 1.8 | 0.59 | 0.09 | 100%

`
or see https://forum.prusaprinters.org/forum/english-forum-general-discussion-announcements-and-releases/minimum-support-distance/#post-152704

Really I just want to know how to make the supports supportive. And I have a STRONG suggestion that the tool tips be updated to reflect that changing the "z contact distance" won't effect the "z contact distance".

As usual, if there is ANYTHING I can do to help this progress, help with the understanding, justification, etc, please let me know.

AbeFM, 2019-07-11:
Original Post:

> ### Behavior
> Supports leave an extra gap. I had this issue well over a year ago, read about #102 #102, and a few others. Since it seemed well understood at the time, I assumed it would get fixed. [...]

bubnikv, 2021-03-18:
To improve quality of the object over support, we have added a new settings: "thick bridges". If enabled (that is by default), PrusaSlicer behaves as before. If disabled, bridges are printed as in any other slicer: The extrusion rate is given by the normal layer height x extrusion width * bridge flow ratio. 

[image]

Implemented with ceea9de8b8b1d4b35bdd1180051033680c405ba1 f01f02154c617cc24835475da419ed6322f2ceec

Duplicate of #341 #4126 #2217

ProTubeVR, 2021-03-19:
> 
> 
> To improve quality of the object over support, we have added a new settings: "thick bridges". If enabled (that is by default), PrusaSlicer behaves as before. If disabled, bridges are printed as in any other slicer: The extrusion rate is given by the normal layer height x extrusion width * bridge flow ratio.
> 
> [image]
> 
> Implemented with [ceea9de](https://github.com/prusa3d/PrusaSlicer/commit/ceea9de8b8b1d4b35bdd1180051033680c405ba1) [f01f021](https://github.com/prusa3d/PrusaSlicer/commit/f01f02154c617cc24835475da419ed6322f2ceec)
> 
> Duplicate of #341 #4126 #2217

Hi,
Is it possible to test this ?
(sorry i don't find the build with it)

bubnikv, 2021-03-19:
Yes, once we have the first public alpha out.

closed 2021-03-23 by bubnikv