prusa3d/PrusaSlicer #2621: Bad right panel layout

horfee, 2019-07-12:
### Version
PrusaSlicer 2.0.0+

### Operating system type + version
MacOS Mojave 10.14.5

### 3D printer brand / version + firmware version (if known)
Prusa MK3S MMU2S

### Behavior
The right panel is slighty cropped on MacOS, in French :
[image]

Then we cannot change support option anymore...
It does happen only when using MMU printer.

YuSanka, 2019-07-13:
Thanks for your report. This bug was fixed and will be a part of upcoming release.

bubnikv, 2019-08-01:
It will be fixed in the upcoming PrusaSlicer 2.1.0-alpha1. Closing.

closed 2019-08-01 by bubnikv