prusa3d/PrusaSlicer #2622: Rafts overlap

Pun-e, 2019-07-12:
### Version
2.0.0 WIN64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa i3 Mk2.5

### Behavior
Have more than one object
Add raft
Majick

[image]

Pretty self-explanatory

bubnikv, 2019-08-01:
known issue, there is an issue open for that.
Closing this one.

closed 2019-08-01 by bubnikv

gretel, 2019-08-22:
could you please refer to the issue you mention to be open.

lukasmatena, 2019-08-22:
E.g. https://github.com/prusa3d/PrusaSlicer/issues/316, https://github.com/prusa3d/PrusaSlicer/issues/1275, https://github.com/prusa3d/PrusaSlicer/issues/684, https://github.com/prusa3d/PrusaSlicer/issues/2598

gretel, 2019-08-22:
@lukasmatena these issues go back to 2017 and 2018. shrug!

gunkl, 2020-04-15:
LOL "Works as designed" - @bubnikv - Can we consider a fix (this could be a feature request, instead of works as designed) - IE, if I want to squeeze as many parts as possible on the print, the raft overlap is always going to be an issue.  Is there a way to add an adjustment that allows rafts to be created with a variable outside of object distance, or a checkbox that allows you to make the raft not generate outside the boundary of the object?

piotrva, 2020-09-10:
I will also welcome such change - it generates issues while printing small elements on rafts and wastes material.

hawken93, 2020-09-18:
I would like to maybe group some objects and create rafts for them as a whole, if that makes sense. Was first thinking of maybe having the slicer compile them into one virtual objects before creating rafts for that, but maybe that would create excessive rafts. I don't know

SeeJayDee, 2021-03-05:
Having seen how TierTime's UP Studio handles rafts for multi part prints, this is more than just a "nice to have" feature and the fact that devs continually shrug it off as unimportant is pretty disappointing.

juanfmx2, 2021-05-13:
Hi guys there is a workaround for this, before slicing you can select multiple parts and merge them into a single one
[image]