prusa3d/PrusaSlicer #2623: FeatureRequest(?): Custom Layer Height for Different Parts on Same Bed

PMARINA, 2019-07-12:
### Version
_PrusaSlicer-2.0.0+win64-201905201652_

### Operating system type + version
_Windows 10 Pro, 10.0.18362 Build 18362_
_https://pastebin.com/L46FuGPT_

### 3D printer brand / version + firmware version (if known)
_FlashForge Creator Pro, Prusa Mk3s MMU2s running 3.7.1_

### Behavior
* Under the `Layers and Perimeters` option for each Part imported (PrusaSlicer recognized that the multiple STLs being imported were parts of a larger object), users are able to select layer heights for each part; however, this is not reflected in the preview (or print time). It seems that PrusaSlicer is using the layer height set under Print Settings>>Layer Height
* _Steps needed to reproduce the problem_
  * Import multiple STLs that form an object
  * When PrusaSlicer asks if these are parts of an object, click 'yes'
  * Set layer heights for the individual parts (that are not the same)
  * Slice
  * Change layer height in Print settings
  * Slice
  * Compare print times, which indicates that the individual layer heights are being ignored
* _Expected Results_
  The objects are sliced with different layer heights, meaning parts that are half the layer height of others require two layers to print, while others do not. 
* _Actual Results_
  See attached screenshots for reference of what I'm attempting. Entire object is sliced with a separate layerheight. 

_Is this a new feature request?_
I believe so. This does not seem to be implemented.
#### Project File (.3MF) where problem occurs (See attached)
[Upload.zip](https://github.com/prusa3d/PrusaSlicer/files/3387492/Upload.zip)
[image]
[image]

lukasmatena, 2019-07-12:
I believe this issue is a duplicate of #2529. If you agree, please, close it.
PrusaSlicer should not have allowed you to set the different layer height.

PMARINA, 2019-07-12:
It looks like it is the same. Can I request it as a feature?

bubnikv, 2019-08-01:
The upcoming PrusaSlicer 2.1.0-alpha1 will support assigning the layer height per object Z range. Does it satisfy your request?

PMARINA, 2019-08-01:
Yes, thank you for including this feature.

closed 2019-08-01 by PMARINA