prusa3d/PrusaSlicer #2624: [MACOS] Error on compile

satiromarra, 2019-07-12:
I tried compile in macOS High Sierra 10.13.6 with this [How to build - Mac OS.md](https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Mac%20OS.md)

Part: Dependencies
```
mkdir build
cd build
cmake ..
make
```
(minutes later)
```
[ 91%] Performing build step for 'dep_wxwidgets'
(multiples warnings)

/bin/sh: msgmerge: command not found
make[4]: *** [af.po] Error 127
/bin/sh: msgmerge: command not found
make[4]: *** [an.po] Error 127
/bin/sh: msgmerge: command not found
make[4]: *** [ar.po] Error 127
/bin/sh: msgmerge: command not found
make[4]: *** [ca.po] Error 127
/bin/sh: msgmerge: command not found
make[4]: *** [ca@valencia.po] Error 127
/bin/sh: msgmerge: command not found
make[4]: *** [cs.po] Error 127
/bin/sh: msgmerge: command not found
make[4]: *** [da.po] Error 127
/bin/sh: msgmerge: command not found
make[4]: *** [de.po] Error 127
/bin/sh: msgmerge: command not found
make[4]: *** [el.po] Error 127
/bin/sh: msgmerge: command not found
make[4]: *** [es.po] Error 127
/bin/sh: msgmerge: command not found
make[4]: *** [eu.po] Error 127
/bin/sh: msgmerge: command not found
make[4]: *** [fi.po] Error 127
/bin/sh: msgmerge: command not found
make[4]: *** [fr.po] Error 127
/bin/sh: msgmerge: command not found
make[4]: *** [gl_ES.po] Error 127
/bin/sh: msgmerge: command not found
make[4]: *** [hi.po] Error 127
/bin/sh: msgmerge: command not found
make[4]: *** [hu.po] Error 127
/bin/sh: msgmerge: command not found
make[4]: *** [id.po] Error 127
/bin/sh: msgmerge: command not found
make[4]: *** [it.po] Error 127
/bin/sh: msgmerge: command not found
make[4]: *** [ja.po] Error 127
/bin/sh: msgmerge: command not found
make[4]: *** [ko_KR.po] Error 127
/bin/sh: msgmerge: command not found
make[4]: *** [lt.po] Error 127
/bin/sh: msgmerge: command not found
make[4]: *** [lv.po] Error 127
/bin/sh: msgmerge: command not found
make[4]: *** [ms.po] Error 127
make[4]: msgfmt: No such file or directory
make[4]: *** [msw/it.mo] Error 1
/bin/sh: msgmerge: command not found
make[4]: *** [nb.po] Error 127
/bin/sh: msgmerge: command not found
make[4]: *** [ne.po] Error 127
/bin/sh: msgmerge: command not found
make[4]: *** [nl.po] Error 127
/bin/sh: msgmerge: command not found
make[4]: *** [pl.po] Error 127
/bin/sh: msgmerge: command not found
make[4]: *** [pt.po] Error 127
make[4]: msgfmt: No such file or directory
make[4]: *** [pt_BR.mo] Error 1
/bin/sh: msgmerge: command not found
make[4]: *** [ro.po] Error 127
make[4]: msgfmt: No such file or directory
make[4]: *** [ru.mo] Error 1
make[4]: msgfmt: No such file or directory
make[4]: *** [sk.mo] Error 1
make[4]: msgfmt: No such file or directory
make[4]: *** [sl.mo] Error 1
make[4]: msgfmt: No such file or directory
make[4]: *** [sq.mo] Error 1
make[4]: msgfmt: No such file or directory
make[4]: *** [sv.mo] Error 1
make[4]: msgfmt: No such file or directory
make[4]: *** [ta.mo] Error 1
make[4]: msgfmt: No such file or directory
make[4]: *** [tr.mo] Error 1
make[4]: msgfmt: No such file or directory
make[4]: *** [uk.mo] Error 1
make[4]: msgfmt: No such file or directory
make[4]: *** [vi.mo] Error 1
make[4]: msgfmt: No such file or directory
make[4]: *** [zh_CN.mo] Error 1
make[4]: msgfmt: No such file or directory
make[4]: *** [zh_TW.mo] Error 1
make[3]: *** [allmo] Error 2
make[2]: *** [dep_wxwidgets-prefix/src/dep_wxwidgets-stamp/dep_wxwidgets-build] Error 2
make[1]: *** [CMakeFiles/dep_wxwidgets.dir/all] Error 2
make: *** [all] Error 2

```

lukasmatena, 2019-07-12:
I'm no expert on this, but make appears to complain about not finding msgmerge and msgfmt, which are part of GNU gettext utilities. wxWidgets use these for [internationalization](https://docs.wxwidgets.org/3.0/overview_i18n.html) and I believe you must have gettext utilities installed in order to compile wxWidgets.

satiromarra, 2019-07-12:
@lukasmatena thanks for reply.
I tried this and work ok:
in command line:
```
ruby -e "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install)" < /dev/null 2> /dev/null
brew install gettext
echo 'export PATH="/usr/local/opt/gettext/bin:$PATH"' >> ~/.bash_profile
```
After closes the terminal and open again.

closed 2019-07-12 by satiromarra

spuder, 2019-12-28:
I am also running into this on the master branch (version_2.2.0-alpha2)

I've ensured `gettext` is installed and is at the front of the path

```
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: /Users/spencerowen/Code/github/PrusaSlicer/deps/build/dep_wxwidgets-prefix/src/dep_wxwidgets/lib/libwx_osx_cocoau_core-3.1.a(corelib_datavgen.o) has no symbols
/bin/sh: PATH=/usr/local/opt/gettext/bin/:/usr/local/opt/gnu-getopt/bin:/usr/local/sbin:/opt/chefdk/bin:/Users/spencerowen/.rvm/gems/ruby-2.5.1/bin:/Users/spencerowen/.rvm/gems/ruby-2.5.1@global/bin:/Users/spencerowen/.rvm/rubies/ruby-2.5.1/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Applications/VMware Fusion.app/Contents/Public:/Applications/Wireshark.app/Contents/MacOS:/usr/local/opt/gnu-getopt/bin:/usr/local/sbin:/opt/chefdk/bin:/Users/spencerowen/.rvm/gems/ruby-2.5.1/bin:/Users/spencerowen/.rvm/gems/ruby-2.5.1@global/bin:/Users/spencerowen/.rvm/rubies/ruby-2.5.1/bin:/Users/spencerowen/.rvm/bin:/Users/spencerowen/go/bin:/Users/spencerowen/Library/Python/3.7/bin:/Users/spencerowen/.rvm/bin:/Users/spencerowen/go/bin:/Users/spencerowen/Library/Python/3.7/bin: No such file or directory
make[2]: *** [dep_wxwidgets-prefix/src/dep_wxwidgets-stamp/dep_wxwidgets-build] Error 127
make[1]: *** [CMakeFiles/dep_wxwidgets.dir/all] Error 2
make: *** [all] Error 2
```

I am still unable to build the dependencies for PrusaSlicer

gdombiak, 2020-02-07:
Hitting exact same error using master commit 455f36119aae14a6a50ed2743b4bb1bb997c8682.