prusa3d/PrusaSlicer #2626: Option to show real filament colors in Plater view

lrconsult, 2019-07-13:
### Version
2.0.0

PrusaSlicer-2.0.0+win64-201905201652

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10 1809
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
MK3+MMU2

### Behavior
It would be great to havbve an optional setting in preferences:
Use filament colors in filement settings instead of extruder colors in printer colors

I think it make more sense to see the color of the assigned filament for each component in the plater view. Now, I would always have to change the printer settings to see the real colors.

_Is this a new feature request?_
Yes

bubnikv, 2019-08-09:
When the multi-material was introduced into Slicer some two years ago, if the extruder color was set to empty string (meaning no color selected, not possible through the UI, but possible by editing the profile files), filament color was used. You may try that.

If it still works, then the only thing missing feature here is really a "reset" button next to the extruder color field, and we may change our profiles to offer this "undefined" extruder color.

twindan, 2019-08-23:
I just checked.  I copied my printer profile to a custom one and then manually editted the .ini file so that "extruder_colour" was empty.  When I loaded Prusa Slicer, the extruder color (in the settings) was black, but it does indeed set the color of the models to the selected filament color. 

As another test, I changed one of the other parameters and saved the profile. The new profile preserved the empty "extruder_color", so indeed, it does seem like the low level functionality is there and the only thing needed is a UI update.

twindan, 2019-08-23:
One other comment on the UI. I started with the default printer profile and then switched to my custom profile. The "extruder color" still showed the previous printer's color (in the UI), even though the 3D view now started using the filament color.

twindan, 2019-08-23:
One more observation. If you change the filament color, it doesn't immediately reflect in the 3D view. You need to add or delete a model, switch to a different tab, switch to the sliced view (and back), or switch to the layers editing and then back.

bubnikv, 2019-08-24:
The upcoming 2.1.0-beta will allow to reset the printer extruder color to "Undefined". In that case the real filament color will be used.

The other bugs should be fixed by the 2.1.0-beta as well. Closing. Feel free to reopen if you find out some of the bugs were not resolved.

closed 2019-08-24 by bubnikv

bivaterl, 2019-09-04:
Not sure if this can go here - can't find where to post about a "closed" item that doesn't have an issue number. You mentioned in 2.1 RC build that you eliminate the first layer color change. Please don't disable! THat is a great setting if you are printing serial models and the last color wasn't the first new color. If you keep the setting, then you can get a color change after the purge strip, but before the first layer prints. I'd prefer it be before the purge strip, but leaving it alone is better than disabling.