prusa3d/PrusaSlicer #2627: Different speed for 1st solid infill on sparse infill

espr14, 2019-07-13:
2.0.0 win64

Solid infill is printed on bed, on solid infill, or on sparse infill. The last one is tricky because it is kind of bridging. Therefore, it should use speed similar to speed of bridges. Moreover, it could use bridge flow ratio.

bubnikv, 2019-08-09:
Thank you for your suggestion. I believe slicer does exactly what you described. Do you think it does not?

espr14, 2019-08-12:
Looks like it does. However, the results in my case are not good. I need to tune the parameters. Still, there could be options for two cases:

1) outside bridges (must look nice)
2) inside bridges (must not fail)

You may close the issue if nobody will implement it soon.

closed 2021-02-02 by rtyr