prusa3d/PrusaSlicer #2628: Missing print host upload feature?

alesti, 2019-07-13:
### Version
2.0.0+

### Operating system type + version
macOS 10.14.5

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S, Firmware 3.7.2_2363

### Behavior
I do not find the print host configuration as in Slic3r 1.41.3+, even if i import my old config (with print host config). 
I do not find any other questions for this in the forum or here - i might search wrong - do i miss something or is this just 'optimized' by you?

I would recommend to get this feature back, it safes some steps.
(using octoprint as print host)

Cheers,
Aleks

lukasmatena, 2019-07-13:
Possibly a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2046 ?

alesti, 2019-07-13:
Dont think so. 
* #2046  is about the old (1.4X) version, where i am able to configure a print host.
* I am able to use a usb serial connection directly to the printer (e.g. to update the firmware).

But i found it now (as i started to make screenshots from the old version to compare and the new one). I needed to switch into advanced or expert mode and then into the common tab.

Sorry,
Aleks

closed 2019-07-13 by alesti