prusa3d/PrusaSlicer #2631: Canceling g-code generation PursaSlicer

dapal-003, 2019-07-13:
### Version
PrusaSlicer-2.0.0

### Operating system type + version
Windows 10 Education, 1903 build 18362.239

### 3D printer brand / version + firmware version (if known)
Ender 3, stock firmware

### Behavior
Problem:
After generating the 3d sliced model with all support prusaSlicer gets stuck on about 90% completion. Once clicking cancel at this point will result in the slicer freezing up for multiple minutes before continuing like normal. 

Procedure:
Import a large STL with difficult to compute parts, 
Generate G-code (Done with supports, and support promoting and blocking boxes)
Cancel the generation after the sliced model is shown and the generating bar is stuck at 90%.

Expected results:
Generation of G-code cancels out and generated results properly handled. 
Disposing the results when the user cancels the generation and allowing the user to save the file if it's completed.

Actual results:
Generating the code goes fine, prusa is showing the bar moving and cpu starts to get loaded. Throughout the whole generation the slicer stays responsive. After a while the process bar gets stuck at 90%, exporting the G-code at this point results nothing happening but the export saying it started. Once clicking "cancel" (before or after exporting) or waiting longer, the whole program freezes up showing a "Not responding" and a white overlay. 
[image]
After multiple minutes the slicer starts to respond again.

Project file:
See attached
[test.zip](https://github.com/prusa3d/PrusaSlicer/files/3389048/test.zip)

userosos, 2019-07-13:
I has the bag too windows 10 ltsc 2019.

dapal-003, 2019-07-14:
After toggling a few options it seems to be "Avoid crossing perimeters" that takes a long time to calculate and can't be cancled.

bubnikv, 2019-08-25:
Aborting issue of the "avoid crossing perimeters" feature has been fixed with 3addae3, and the feature is now also significantly faster.

The G-code generation step now takes around 5 minutes on my computer with "avoid crossing perimeters" enabled, it took some 30 seconds with the "avoid crossing perimeters" feature disabled.

The estimated print time is 2d1h9s with the "avoid crossing perimeters" enabled, 1d0h52m with the "avoid crossing perimeters" feature disabled. That's a good thing, it does not add too much printing time.

The G-code can now be canceled. Closing.

closed 2019-08-25 by bubnikv