prusa3d/PrusaSlicer #2632: Allow infil percentage in purge block.

BarnumD, 2019-07-14:
### Version
2.0.0

### 3D printer brand / version + firmware version (if known)
Prusa Mk3S MMU2 (Upgraded from MK3)

### Behavior
The printer crashes into hills/mountains on the purge block - creating misalignment on the main object.  When purging to the purge block, too much material sometimes extrudes, creating a hill.  On subsequent runs, more and more material is added, creating a small mountain of material.  Eventually the printer crashes into this mountain which messes up the alignment of the printer.  I believe the main problem here (aside from perhaps tuning filament settings) is that since the purge block seems to use 100% infill there is nowhere for the material to go but up.

_Is this a new feature request?_
Yes - Allow purge block infill percentage setting.  I believe that allowing an infill percentage on the purge block would allow the user to program a variable amount of gap between purging lines, thus allowing any extra material to squeeze into the gaps.

rtyr, 2019-07-16:
The behaviour you are describing happens when the MMU2 (without S) printer profile is being used on the MMU2S printer. Are you sure that you are using the correct one (Original Prusa i3 MK3S MMU2S)? You can also check your IR sensor calibration (https://help.prusa3d.com/article/wprwemqsmj-ir-filament-sensor-calibration).

BarnumD, 2019-07-17:
I checked and I may have had both printer profiles installed.  So i re-ran the slicer configuration to only install the Mk3S MMu2S profile.  I found the instructions for MK3s confusing to find, and I never saw the instructions for calibrating the IR sensor but I will try that now.

BarnumD, 2019-07-21:
I believe your assessment was correct.  After ensuring the correct printer profile was used and prpoerly calibrating the IR sensor calibration, the purge block has not created this type of issue again and there seems to be enough gaps in between to handle this sort of thing.  Additionally, the IR calibration helped to avoid MMU load failures.

closed 2019-07-21 by BarnumD