prusa3d/PrusaSlicer #2635: [Feature Request] MMU - Purge to Supports

MrFlippant, 2019-07-14:
I searched, but was unable to find this suggestion posted already. Please forgive me if it has been.

For MMU printing, I noticed that supports are simply printed using the last loaded filament for each layer, rather than as any part of the purge/color change process. I'm currently printing a 4 color print that requires a lot of support that could easily be used to continue the color change/purge process. I hate waste, so I have purge/waste objects on the plate to minimize the purge tower size. Sometimes it's hard to find good objects that can be made to fit the primary object and serve as useful purge volume, so it would be very nice to have the option of using support structure to continue that process. 

There may need to be a note about min purge to tower. I've seen some profiles where it has been set very low, such as 5cmm, but the default of 15cmm seems to be pretty reliable to prime the nozzle. Support would definitely need to be printed with a properly primed nozzle, but there's usually a good 50+cmm that needs to be extruded for proper color change.

lukasmatena, 2019-07-15:
This is more complicated than it looks, because in general case, the supports are not on the same layers as the objects. They are allowed to be "in between".  The extruders are always sorted so that the last extruder on one layer is used as the first extruder on the next, which implies that there is no need to purge anything when the supports are printed (as long as both layers use the same extruders, which they usually do).

This means the feature would only be useful in case that the object and support layers are synchronized. And given how much work it would mean for this limited benefit, I'm not sure when (if at all) we will implement it.

MrFlippant, 2019-07-15:
Makes sense. One can always hope. 
I should have thought about this earlier in my current print... I was running low on one of the colors, but thought it would be enough... but it's one of the last two colors, so is used for supports half the time, and is going to run out before the print is over. :(
From now on, colors with the most remaining will go in the last two slots for any given print.

verner33ob, 2019-07-19:
I think, there could be not so complicated way, to calculate for the "in between" problem.
For example calculate the next "part layer" first, then calculate the "support layer", so it would alrady know, if a purge needs to happen or not.
Or just check, what color the next "part layer" will start with. Again, then it would be known, if a priming is needed, before the supports are generated.

lukasmatena, 2019-07-19:
The problem is not about calculating whether there is something to purge. The problem is that in most cases there is not. Usually layers use the same set of extruders as the previous layer, so the extruder must have been purged before the previous layer was finished. And I would prefer not to change the whole gcode generation logic because of the few layer where a bit of material could be saved.

verner33ob, 2019-07-19:
Yes, you are right. My bad. I read over the fact, that the last extruder and the first extruder are the same, if possible. So obviously there wouldn't be the need for purging. And only on big objects, it woul eliminate the purge tower anyway, so yes, unfortunately "purge to supports" is at the moment no option for MMU-Extruder.

jwillmer, 2019-07-29:
same as #2671 ?

MrFlippant, 2019-07-29:
No, that's for infill, not supports. Two different things. I already use wipe to object and infill. It would just be nice to use supports for wiping as well. 

I did just notice an option in slicer to "Synchronize (supports) with object layers" but that appears to be only allowed/used for soluble supports. Still, it's in there, so maybe it can be used somehow. The interface layers would probably still need to be "between" layers to provide good support without solid adhesion when using regular PLA for all materials, though.

Still hoping. :)

NerdBrez, 2020-04-10:
> ...The extruders are always sorted so that the last extruder on one layer is used as the first extruder on the next...

That is not happening. In my test (file attached), in the 2nd layer the 1st extruder should have been the last selected so it would continue the rest of the print without anymore changes. It is being selected first, what forces a new tool change in the beginning of the 3rd layer.

[example.zip](https://github.com/prusa3d/PrusaSlicer/files/4463568/example.zip)

MrFlippant, 2020-04-10:
In your file, the layers would be printed as follows:
Layer 1: E1, E2, E3, E4, E5
Layer 2: E5, E1, E2, E3, E4 (last layer of all but one object). Since E4 is not used on the next layer, there must be a tool change back to E1 for the remaining layers.
Layer 3+: E1

I think you're thinking about it wrong. PrusSlicer isn't going to re-arrange the extruder order so that the last tool changed-to prior to the first layer of the rest of the object is the same one needed for the rest of the object. He's saying that the last tool used on any given layer, regardless of what tool that is, is the first tool being used on the next layer... IF that tool is NEEDED on the next layer.

If your other objects were a few layers taller, you would see the following pattern repeat:
E12345, E51234, E41235, E51234, E41235.. and so on.

The reason I started this request was because support material, which is already a waste product, is not being used as purge material. For a two color object with supports, being able to do that would be fantastic. For more colors, it would be less useful, but still better than not using it at all.

NerdBrez, 2020-04-13:
@MrFlippant I got it, thanks for explaining. 
@lukasmatena I think the algorithm should check what extruders are used in next layer too before setting the extruder order for the current layer. That would be a slight improvement and I wonder if it would be hard to implement. It is important to get rid all possible tool changes and depending on the model this could gain some print time and avoid some waste.

MrFlippant, 2020-04-13:
Probably want to start a new request for that.

closed 2026-09-29 by prusaslicer-bot[bot]