prusa3d/PrusaSlicer #2636: [Request] What about dev build?

userosos, 2019-07-14:
Dev build will be released?

lukasmatena, 2019-07-15:
Only master branch builds are released and only in well defined stages of development. If you want to see a state of a dev branch, you have to compile it yourself (the code is public). Hopefully this answers the question, so I'm closing the issue.

closed 2019-07-15 by lukasmatena

userosos, 2019-07-15:
I think that the repo have the dev builds.
https://github.com/supermerill/Slic3r

lukasmatena, 2019-07-16:
Ok, you're most likely asking about whether we are going to start using some continuous integration tool. Similar question was asked in https://github.com/prusa3d/PrusaSlicer/issues/2037, I'd suggest to discuss it there. A short answer would be that there is no public CI at the moment and there is a lot of other work to do.