prusa3d/PrusaSlicer #2639: Supported G-Code List?

TNDavid, 2019-07-15:
I have searched the web for a correct list of G-Codes used and supported by Prusa Slicer, but have been unsuccessful.  Where can I find an up to date listing with descriptions?

bubnikv, 2019-07-16:
https://github.com/prusa3d/PrusaSlicer/wiki/Slic3r-Prusa-Edition-Macro-Language
https://github.com/prusa3d/PrusaSlicer/wiki/Slic3r-placeholders-(a-copy-of-the-mauk.cc-page)

bubnikv, 2019-08-01:
Answered, closing.

closed 2019-08-01 by bubnikv

TNDavid, 2019-08-02:
Why closed?  I found no list of supported G-codes!  I am looking for the list of G***, M****, etc. Cannot find that list anywhere?

bubnikv, 2019-08-05:
https://en.wikipedia.org/wiki/G-code
https://reprap.org/wiki/G-code

Please understand that it is not very economical if a software architect performs the role of user support.