prusa3d/PrusaSlicer #2640: Linux missing permission to flash firmware

theaquarium, 2019-07-15:
### Version
_Version of PrusaSlicer used goes here_
2.0.0+linux64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
Linux 5.0.0-15-generic
Pop!_OS 19.04

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Stock Prusa i3 MK3 / Firmware 3.7.2

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
1. Open the PrusaSlicer AppImage (no root access)
2. Open the Firmware Flash dialog
3. Select firmware and port and press flash

The dialog says "Flashing failed. Please see avrdude log below." In the log, avrdude attempts to open the port and fails due to lack of permission. This can be fixed by running PrusaSlicer with sudo. 
[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

theaquarium, 2019-07-15:
BalenaEtcher for Linux opens a password dialog when you flash an SD Card/USB, so maybe this could be recreated.

lukasmatena, 2019-07-15:
What's wrong? What do you expect PrusaSlicer to do?

vojtechkral, 2019-07-15:
@theaquarium Most likely you need to add your user to the `dialout` group. Cf. [here](https://askubuntu.com/questions/112568/how-do-i-allow-a-non-default-user-to-use-serial-device-ttyusb0).

We could also probably do something similar to those etchers (ie. run some small data-shovelling program such as `dd` or `cat` with root privileges and use it to pipe data to the device), but I'm not sure this is worth the cycles at the moment...

bubnikv, 2019-08-01:
> BalenaEtcher for Linux opens a password dialog when you flash an SD Card/USB, so maybe this could be recreated.

I cannot imagine how an application, which does not have super user rights could ask for COM port permission. You have to give the permission to Slicer yourself, sorry.

Closing.

closed 2019-08-01 by bubnikv

theaquarium, 2019-08-02:
Would it be possible to attempt to run the avrdude with admin privileges and have the system prompt the user (like with a password or UAC dialog).

bubnikv, 2019-08-02:
yes, but our firmware updater is currently integrated into PrusaSlicer and
it is heavily modified. We would have to compile it as a separate
application and somehow give it the sudo priviledges. This is non-trivial.

I believe it is much easier just to let our software to explain the user on
Linux what to do (to put PrusaSlicer into the group that has access to
serial ports). IMHO if a Linux user uses Linux, he should be qualified
enough to understand that upfront, but unfortunately this is not the case.

vojtechkral, 2019-08-02:
> Would it be possible to attempt to run the avrdude with admin privileges and have the system prompt the user (like with a password or UAC dialog)?

In theory, yes, but I don't want to do this. IMO giving `avrdude` unlimited privileges is not a good idea. Our case is different from those SD card / USB flash etchers - all they need to do is open `dd of=/dev/something` with sudo, pipe into that and call it a day. avrdude, on the other hand, needs to read, write, and perform device-specific ioctls. Those etchers also need write permissions to arbitrary block devices, but we only need to write to serial and we _don't_ want write permissions to arbitrary devices and system files. I'm certainly not keen on users attempting to flash firmware onto their `/dev/sda` or something (I know it seems unlikely that someone would even attempt that, but users are sometimes surprisingly creative).

Adding your user to the appropriate group is recommended by distros, it's a solution that makes sense and is safer, so we're sticking to that. We can certainly improve the UX, perhaps better explain to the user what they need to do, we could also read the group id from the device and tell the user. We'll do that if/when we have enough spare time/cycles.

theaquarium, 2019-08-02:
Yeah, I didn't consider that you could do that. I suppose a dialog explaining how to do this would work. Thanks for the help.

panasenco, 2023-06-24:
Hello, I'm also on Pop!_OS and this is how I resolved this:

1. Add my user to the "dialout" group with `sudo adduser $USER dialout`.
2. Restart my computer to make the group changes take effect.
3. Launch PrusaSlicer with access to /dev enabled with `flatpak run --device=all com.prusa3d.PrusaSlicer`
4. Firmware flash now proceeds without issues.