prusa3d/PrusaSlicer #2643: Impractical Reslicing

uacajala, 2019-07-16:
### Version
PrusaSlicer
Version:   2.0.0+win64
Build:     PrusaSlicer-2.0.0+win64-201905201652

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 17763), 64-bit edition

OpenGL installation
GL version:   4.5.0 NVIDIA 377.83
Vendor:       NVIDIA Corporation
Renderer:     Quadro 2000/PCIe/SSE2
GLSL version: 4.50 NVIDIA

### Operating system type + version
Windows Professional 10

### 3D printer brand / version + firmware version (if known)
MK3S

### Behavior

Small changes result in a complete reslice of the model. For example when I change the filament price the model reslices in order to give me that information. I think that a small change like that should have a way to use the existing slicing data to recalculate without having to completely reprocess.

bubnikv, 2019-07-17:
I agree that the filament price should not reslice the object.

PrusaSlicer implements a complex caching to avoid such recalculations. Do
you known about more of such parameters that you believe trigger
unnecessary recalculation?

bubnikv, 2020-02-04:
Sorry, but there is a reason why a change of any parameter reslices: The G-code exporter does not know whether there is any reference of any config key in your custom G-codes.
Anyway, I don't think it is such an issue. One does not change the filament price often.

Closing.

closed 2020-02-04 by bubnikv