prusa3d/PrusaSlicer #2644: Top layer is actually printed as solid infill on some models

ManuGithubSteam, 2019-07-16:
### Version
2.0.0

### Operating system type + version
Arch Linux

### 3D printer brand / version + firmware version (if known)
Mk3s

### Behavior
* _Describe the problem_

With the attached STL File, the slicer does slice it correctly and does mark the top layers around the E as top layers. Hower the layers are actually printed as Solid infill and not as top layers.

For comparison: Slice a normal cube without the pulled in E and print it. It is printed with top layers on the top side and not with solid infill. The solid infill is only there because of the pressed in E.

* _Steps needed to reproduce the problem_
Slice the stl and print it. Watch out for the solid layer speed and the top layer speed. Its easy to spot if those are very different

* _Expected Results_
Top layer should be printed as top layer even when the geonetry is like this

* _Actual Results_
Top layer is actually solid infill

[TEVO_Cube_FormFutura APOLLOX ASA - 0.25mm QUALITY - MK3S 0.4 v3.zip](https://github.com/prusa3d/PrusaSlicer/files/3398708/TEVO_Cube_FormFutura.APOLLOX.ASA.-.0.25mm.QUALITY.-.MK3S.0.4.v3.zip)

[top_layer.zip](https://github.com/prusa3d/PrusaSlicer/files/3398702/top_layer.zip)

MR2C280, 2019-07-17:
Could this be why I'm getting this rippling.  It looks like overextruding that isn't attaching to the layer below (printed using a new spool of Prusament)?

[image]

Only the "0.35mm FAST" default Print setting has such a big difference between the default extrution width and the Solid infill.  Is it correct?

[image]

ManuGithubSteam, 2019-07-17:
Hi @MR2C280  I dont think your issue is related as the slicer slices and uses a normal shape like a rectangle in the correct way. just special geometry like the pressed in E in my stl are affected so far. I printed a flat test cube to confirm this.

My guess for your issue is: your bed is not level. If you use octoprint please install the plugin bed visualizer and use these g-codes in it:
G80
@BEDLEVELVISUALIZER
G81

Also flip y axis the get right orientation. See if your bed is level.

MR2C280, 2019-07-17:
It's got nothing to do with bed level - I'm using an Original Prusa MK2.5S with mesh bed leveling (and I don't use octoprint).  The first 4 solid layers print fine (and other smaller prints are fine too).  Only when it comes to the top solid layer does it go weird.  I suspect it's using the "Top solid infill" width for slicing but using the "Solid infill" width for extruding volume.  I'll try it tomorrow with "Solid infill" set to 0.43.  If that doesn't work I'll try your idea of changing the orientation.

ManuGithubSteam, 2019-07-17:
If you are correct then it would be the same issue as mine. Could you print the tevo cube from my zip file and post a picture ?

bubnikv, 2019-08-09:
@ManuGithubSteam I loaded config from you G-code and I sliced your two models. Top layers are sliced with top infill, at least that's what slicer thinks.

[image]

The solid infill is shown with magenta by slicer. I had to slide down to uncover the solid infill.
[image]

bubnikv, 2019-08-09:
Would you please attach the 3MF for which you believe slicer produces incorrect G-code?

manukieli, 2019-10-17:
I've the sale issue with layer 0,15 and top solid to 70%
[image]

rtyr, 2021-02-03:
No response. It is sliced correctly with 2.3.0 using provided STL and settings from provided gcode. Closing.

closed 2021-02-03 by rtyr