prusa3d/PrusaSlicer #2645: Blue Screen  AND  Program resident in task manager after closed

cradelman, 2019-07-16:
### Version
_Version of PrusaSlicer used goes here_
PrusaSlicer-2.0.0

_Use `About->About PrusaSlicer` for release versions_
201905201652

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
Windows 7 Pro x64 Version 6.1.7601

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Prusa MK3

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
Since uninstalling the older Prusa software and installing the May 2019 version, I have experienced 2 issues.
1.  After closing the PrusaSlicer-2.0.0 software, the program does not really close but remains resident in task manager.  Each time i open it, one additional "prusa-slicer.exe *32" remains in task manager until the computer decides to freeze then blue screen.  I can manually kill the processes but this seems to be a nuisance
2.  After running large slicing projects, the computer will freeze, then blue screen.  This doesn't happen all the time, but having my computer reboot is a problem

NOTE:  The slicer is working great and super fast, but seems to not be closing gracefully and crashing my computer periodically

_Is this a new feature request?_
Yes

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
1. remaining program running on windows task manager happens every single time
2. Intermittent rebooting using STL files from thingiverse.  Seems to be larger files that result in blue screen

Thanks

bubnikv, 2019-07-17:
If you get a blue screen of death, then it is most likely an issue of your
graphics driver. Please update your graphics driver.

cradelman, 2019-07-19:
Ran Windows update driver and searched the internet; says the latest is installed for my AMD Radeon HD 6450.

Still having issues with Slicer not closing gracefully and leaving open files in Task Manager

bubnikv, 2019-08-09:
One of our colleges experiences crash of the application on closing. He also runs on Windows 7. We will try to install our development suite on his computer and investigate the issue.

Also our developer @YuSanka has a Windows 7 computer at home, she will try to reproduce the issue there.

cradelman, 2019-08-09:
Do they also see multiple instances running in task manager that have to be manually closed?  Mine always have to be manually “end tasked” in order to close them and recover the used RAM.  The program GUI closes but something is left running in background that isn’t closing

 

Thanks

josedpedroso, 2019-08-16:
I also get lingering prusa-slicer.exe processes after closing the program. Running PrusaSlicer 2.0 on Windows 7 x64.

bubnikv, 2019-08-24:
We were able to reproduce crash on closing PrusaSlicer 2.0.0 and 2.1.0-alpha1 on one of RnD member's machine. We fixed that at least for him. We hope that with PrusaSlicer 2.1.0-beta release your problems will be gone. Closing.

closed 2019-08-24 by bubnikv

tjcolejr, 2019-08-24:
I had this same issue in Windows 10 and you said at the time that nobody else was experiencing it and we concluded that it may have had something to do with my graphics card because it only happened when I was using 2 screens.  2.1.0-beta seems to have fixed my issue as well!  Hooray!

bubnikv, 2019-08-25:
@tjcolejr thanks for heads up