prusa3d/PrusaSlicer #2648: Extrusion Multiplier not working?

BhagavadGitHub, 2019-07-17:
2.0.0+win64 Version
Windows 10 Operating system type
i3 MK3s 3D printer 

I can change the extrusion multiplier and the filament density, but the filament flowrate command stays the same in the g code:

M221 S95

Which to my understanding is 95% flowrate. I am new to g code so maybe I have a fundamental misunderstanding. When I manually change the line to M221 S84 I get much better prints. I just want to adjust it with the slicer instead of manually going into the g code.

BhagavadGitHub, 2019-07-17:
Nevermind, I think this is the issue: https://github.com/prusa3d/PrusaSlicer/issues/1446

bubnikv, 2019-08-09:
Duplicate of #1446. Closing this one.

> When I manually change the line to M221 S84 I get much better prints. I just want to adjust it with the slicer instead of manually going into the g code.

You can do both, either one will work, though flow rate of 84% indicates either some weird filament or somehow wrong calibration of your extruder steps. You may want to contact Prusa3D support.

closed 2019-08-09 by bubnikv