prusa3d/PrusaSlicer #2654: Wipe tower not generated when enabled

uorbe001, 2019-07-19:
### Version 2.0.0+-201905201652

### Operating system type + version
Operating System: Macintosh
System Architecture: 64 bit
System Version: macOS High Sierra Version 10.13.6 (Build 17G7024)

### Behavior
* Having wipe tower enabled, wipe tower is not generated.
[image]
* Slice with the provided 3mf file
  * Wipe tower enabled
* _Expected Results_
  * Wipe tower is generated on the output gcode.
* _Actual Results_
  * No wipe tower
[image]
[image]

#### Project File (.3MF) where problem occurs
[wipe tower.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3410741/wipe.tower.3mf.zip)

uorbe001, 2019-07-19:
As visible on the screenshots, the wipe tower is not even on the preview window at all, and then it's not generated on the gcode.

bubnikv, 2019-07-19:
Currently the wipe tower is only generated for "single extruder multi
material" printer profiles.
The upcoming 2.1 will support wipe towers for dual extruder setups as well.

uorbe001, 2019-07-19:
Ah, that sucks. Any chance the development builds have it?

uorbe001, 2019-07-19:
I'll close it as it's already answered

closed 2019-07-19 by uorbe001