prusa3d/PrusaSlicer #2655: Application Crashes when adding infill via right click

kjmoore, 2019-07-19:
### Version
PrusaSlicer 2.0.0+win64+201905201652

### Operating system type + version
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 18362), 64-bit edition

### 3D printer brand / version + firmware version (if known)
Prusa MK3S + MMU2S

### Behavior
Adding infill to multiple objects can cause the application to close without warning.

Steps:
* Add multiple objects to the bed
* Click away so nothing is selected
* Right click on an object and add infill
* Right click on a second object and add infill
* Application closes

Workaround
If the second (and subsequent) objects are selected with a left click before right clicking to add infill, this error does not occur.
[right-click-crash-prusaslicer.zip](https://github.com/prusa3d/PrusaSlicer/files/3412111/right-click-crash-prusaslicer.zip)

lukasmatena, 2019-07-19:
I managed to reproduce the crash in released PrusaSlicer 2.0.0. Current master seems to be ok. I bet @YuSanka has already fixed it.

bubnikv, 2019-08-01:
This issue will be fixed by the upcoming PrusaSlicer 2.1.0-alpha1. Closing.

closed 2019-08-01 by bubnikv