prusa3d/PrusaSlicer #2656: PrusaControl can't properly arrange 60 copies of my object on the bed

MisterGlass, 2019-07-19:
### Version
Version:   2.0.0+win64
Build:     PrusaSlicer-2.0.0+win64-201905201652

### Operating system type + version
Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 17134), 64-bit edition

### 3D printer brand / version + firmware version (if known)
Prusa I3 MK3S

### Behavior
I am making a batch of a whole lots of the same small item. PrusaSlicer can arrange 59 of them just fine, but it wont place the 60th on the bed - even though there is crealy enough foor for at least one more copy (see attached photo)
[image]

I am able to arrange the item manually and print fine (well, its going now, but seems to be printing fine).

#### Project File (.3MF) where problem occurs
[prusacontrol.error.zip](https://github.com/prusa3d/PrusaSlicer/files/3412677/prusacontrol.error.zip)

tamasmeszaros, 2019-07-22:
Hello and thank you for the report. I think it is fixed now in one of our upcoming feature branches so the next release should not have this issue.

bubnikv, 2019-08-01:
Fixed in the upcoming PrusaSlicer 2.1.0-alpha1. Closing.

closed 2019-08-01 by bubnikv