prusa3d/PrusaSlicer #2657: [Feature Request] Remove "Extruder" option from modifier meshes for MMU

cjbaar, 2019-07-19:
### Version
2.0.0

### Operating system type + version
OSX

### 3D printer brand / version + firmware version (if known)
MK3S + MMU2S

### Behavior
The new version of PrusaSlicer has a lot of great improvements, but it is still missing one feature I was hoping for, which is the ability to use modifier meshes on MMU prints. Every time a modifier is created or added, it automatically gets an "Extruder" parameter added to it that cannot be removed. When trying to make MMU prints, this makes the modifier functions effectively useless, but I can't apply a modifier to a portion of the model without also changing its color (extruder).

#### Project File (.3MF) where problem occurs
Attaching a project file that shows 5 MMU blocks, with a cylinder modifier. The modifier overrides the color where it crosses the print.
[mmu2test.3mf.gz](https://github.com/prusa3d/PrusaSlicer/files/3412897/mmu2test.3mf.gz)

bubnikv, 2019-07-22:
I see. The user interface supports the "Default" extruder, but the
"Default" extruder defaults to the extruder assigned to the object, not to
the extruder assigned to the printable volume that the modifier volume
overrides. I believe the issue is clear, there is no need for user
interface change, but the slicing core logic needs to be modified, which
will take some thinking.

cjbaar, 2019-07-25:
Thanks @bubnikv. Appreciate you taking a look at it.

bubnikv, 2019-09-19:
It is a general issue of how the modifier mesh parameters are processed. Currently the modifier meshes override object parameters, not the part parameters that they overlap with. Unfortunately the slicing back end invalidation mechanism is tightly coupled to the current parameter inheritance hierarchy, it will take quite some effort to change it.

cjbaar, 2019-09-19:
@bubnikv Understood. Thanks for your time in looking into it. I think this would definitely be useful for MMU prints.

bubnikv, 2021-06-03:
> It is a general issue of how the modifier mesh parameters are processed. Currently the modifier meshes override object parameters, not the part parameters that they overlap with. Unfortunately the [...]

It took me couple of sweaty weeks actually, the refactoring has been done also to support the new MM painting feature.

In PrusaSlicer 2.4.0-alpha1 the modifier volumes are applied strictly in the order they are presented at the side panel, one modifier may override the other and a "default" extruder means "don't change, use the parent's".

closed 2021-06-03 by bubnikv