prusa3d/PrusaSlicer #2658: Changing printer from mk3mmu2s to mk3s not processed properly - no spiral

pntucker2, 2019-07-19:
### Version
2.0.0+-201905201652

### Operating system type + version
OSX 10.14.5
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
n/a

### Behavior
I cleared an existing item from the screen in the slicer
I then added a STL file and changed the print setting to .15 and changed the printer to the mk3s - no mmu
I changes the print settings to spiral

* _Expected Results_
the
[retro-bowl.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3412990/retro-bowl.3mf.zip)
[image]

 file would be sliced
* _Actual Results_
see screen shot
_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2019-08-09:
Print.cpp:1147
        if (m_regions.size() > 1)
            return L("The Spiral Vase option can only be used when printing single material objects.");

I suppose the condition may fail, but I wonder why. Are you able to reproduce the issue yourself?

bubnikv, 2020-02-08:
This issue has already been fixed in 2.1.0-beta with ded2019765643d6a0acfe1c296eb1f2d2e860eb9.
Closing.

closed 2020-02-08 by bubnikv