prusa3d/PrusaSlicer #2660: Trying to define a support enforcer box is hopelessly counter intuitive. Scales in all axes change when you alter one of them and overall it makes little sense.

rogerfroud, 2019-07-21:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

neophyl, 2019-07-21:
You don't even list what version of slicer you are using, what your OS is or precisely what your issue is.  Your entire description is very vague.  Personally I find manipulating the support modifiers much better than they used to be in the old version.

some common things to be aware of
a) scale listed in the text boxes on the right are by default locked to all be the same.  You need to deselect the orange padlock icon to make them individually changeable.  This is the same for both model parts and modifiers.
b) its very very easy to just drag and resize the new modifiers and also reposition them by dragging compared to the old way.  You can use the icons on the left or use the keyboard shortcuts once you have selected the part/modifier you want to modify (M for move, S for scale, R for rotate etc there is a list under help).

Like any piece of software you have to spend some time learning how to use it to make best use of it.

cggorman, 2019-07-21:
Agreed. The new features are EXTREMELY welcome.  I do 99% technical/functional parts and use supports on most.

I do wish the actual generated supports more closely matched the bounding shape. I have do a lot of back & forth scale/slice to get exactly what I want.

Would also love the ability to nudge the bounding shapes in tiny increments with the arrow keys.

rogerfroud, 2019-07-21:
neophyl - I'm using PrusaSlicer 2.0.0, I'm hardly likely to be talking about an old version when this is a new feature. Just because you find it better than an old version doesn't mean it's much good!

a) Why? That makes no sense and why I assumed this was a bug!

b) No it isn't, Moving the box can end up in it moving in 3 axes at once which is very confusing. Why would you ever want to do that? If you're looking at it in one orientation and move the box, it can zoom miles away in another axis. This is NOT intuitive.

I'm happy to learn how to use software, and I write the stuff myself, so I'm well aware of the issues. Don't be so defensive when there are justifiable criticisms of your pet software. Hopefully the developers CAN accept criticism and improve the product.

Intuitive interfaces need next to no learning. The fact that you have to figure out how it works because the things you intuitively try don't work, means it's NOT intuitive and can be improved!

cggorman, 2019-07-21:
Simply use the appropriate handle to scale/rotate/move.  You can select a handle to scale in one direction or all, move along one axis or freely, and rotate with or without snaps in any one axis at a time.

Very intuitive if you've ever used 3D software.

rogerfroud, 2019-07-21:
I use Alibre Design 3D software, I'm not new to this sort of thing. It's not intuitive if weird things happen such as the part zooming off to one side when you move it!

neophyl, 2019-07-21:
If you write software then you should appreciate that bug reports need to be specific and helpful. 

Even with your further description it is hard to tell what you mean by zooming around. I use Prusa slicer weekly for my projects but I have not encountered what you describe. Do you have a sequence of steps we can follow that would illustrate your frustration with how the software currently works ?

As for the point A not making any sense it’s actually common for a lot of software, for example various image editing programs have the scale locked to keep perspective, what makes no sense to you does make sense to many other people. I do think that user documentation could be vastly improved as the learning curve is steep and it’s all trail and error. Some decent documentation would go a long way to making it more usable. 

The modifiers did exist in the old software but were much harder to even find, let along use so asking what version you are using is not a pointless question, they are not a new feature at all. 

I am just a happy user of the software, could it be improved, sure what software can’t. However you have to take everyone’s use into account, what one person thinks needs changing another will like just the way it is so developers have to take a careful and measured approach to changes. While I don’t write software for a living I do test it so I guess I appreciate a clear and concise bug report more than most.

cggorman, 2019-07-21:
Certified Professional in Autodesk Inventor, here.  Perfectly intuitive to me, tho I do wish PSlicer would allow pan and zoom to cursor.

rogerfroud, 2019-07-21:
neophyl - It was more of a general moan than a specific bug report, but everyone seems to have understood the changing of all of the sizes happens and that's one thing I was referring to. Yes, image software changes them together, but that's makes sense when you want to maintain an aspect ratio.
It makes no sense here though, and I can't believe anyone would think that changing the X dimension of a box ought to change the Y and Z! That's counter intuitive, and makes my point nicely.

How much time does anyone spend reading documentation? The point of making things intuitive is so that you do less of that. If you have to look things up, then it's pretty likely that it's not obvious how it works. Just because you find it obvious, doesn't mean that everyone else does.

For the other gripe, just grab hold of the box and move it. It moves in 3 dimensional space, all axes at once. What's the point of that? You move it in one view to where it looks sensible, only to find that it's shot off miles away in another axis when you view the build platform from another angle. Maybe that's what you expect, but it isn't what I would and it's not useful behaviour.

neophyl, 2019-07-21:
Ah don’t just grab the box.  Select the box, then either select the move icon from the right hand controls or press M. That should bring up the direction arrows. If you grab the arrow for the axis you want to move it is constrained to that axis. It works a lot like blender. 

Same with rotate, if you grab an axis it just rotated in that axis, same with scale too.  Once you understand how it works it’s VERY easy to use and very quick. 

You do realise your argument about not being intuitive can be turned around 180 degrees ?  It was very intuitive to me as I’m used to that from other bits of software, it’s all personal perspective.  My point about documentation is that at the moment it doesn’t exist in any form that I can find and no matter how intuitive something is it’s clearly needed as your issue proves, there is no such thing as obvious.  Not when people are part of the equation anyway lol.   A good manual is the fall back position for when you can’t just figure it out, without that safety net people get frustrated.

The scale lock is global, when you consider that it also applies to your imported model stl file as well as the modifiers for that model then it does make sense as most users scaling their models don’t usually want to alter the proportions.  If you unlock it then it is unlocked for both.    Now if the Prusa team hadn’t allowed the unlocking then that would be a major bug but this is more that you don’t agree with the default behaviour.   In this case if I were to agree with that view I would raise a feature request for the default behaviour to be configurable in the settings as it’s not a bug but a enhancement.

JohnOCFII, 2019-07-21:
For those interested, I found this short video very helpful to learn the basics on support blockers and support enforcers in PrusaSlicer: https://www.youtube.com/watch?v=yqMXfvWA-aY&t=25s

rogerfroud, 2019-07-22:
neophyl - So DON'T do what you DO do to move models around the bed? Inconsistencies like this are confusing and counter intuitive. If you're not supposed to move the boxes that way, remove that function so it then become clear that this ISN'T the way it's supposed to be done.

The argument isn't turned round 180 degrees when the very same software works in different ways. You just happen to use other software that works in another way and you're ignoring the fact that the slicer is inconsistent.

The problems are due to software that was intended for moving and managing models was pressed into service for something it wasn't intended for with little thought as to whether there were confusing consequences. 

Hopefully they will now modify this confusing behaviour so others don't have the same frustrating experience I had.

bubnikv, 2019-07-22:
@rogerfroud I have a hard time to follow your complaint, but it certainly is not hopelessly counter intuitive if you are the only one complaining about it.

I suppose you are complaining about panning of the modifier mesh in the plane parallel to the print bed. I agree it may be confusing to some extent, but it is consistent with panning of an object along the print bed. So if we just suppress the panning of the modifier meshes, we would make the behavior inconsistent with the panning of the objects. I suppose one shoe will not fit all the feet.

rogerfroud, 2019-07-22:
bubnikv - Other people not complaining simply means they haven't complained... nothing more than that. It's not an endorsement of how good the software is.

I'm pointing out that you are encouraged to grab a model on the bed and move it about, so that's what I tried with the enforcer box. The behaviour isn't the same as you claim, because the part is kept in contact with the build platform and just moves in X & Y.
However, that doesn't allow the enforcer box to move in the Z axis, so the behaviour has changed to allow free movement in Z too. This results in it being frankly useless for moving the box. Try putting the box where you want it to be like that and you will see what I'm complaining about. This is NOT consistent with the movement of a part on the bed.

If they are going to allow you to move the part in all three axes, it would make more sense to arrange for it to only move on one plane at a time, say X/Y X/Z or Y/Z depending on the angle you were looking at it. As it is, you think you're moving it somewhere only to find it's nowhere near where it appeared to be.
This is what I described as hopelessly counter intuitive, and I stand by that.

It you're not intended to move the enforcer box like that, then disable it.

TheDuggem, 2019-07-23:
I had a little trouble with this too until now.  I even watched the Chris Warkocki video linked above when it first came out but I somehow didn't notice what he was doing to move and scale the modifiers.  In all fairness, it was late, I was in bed and watching on a phone instead of the computer.

Dragging the modifier around does move the object off in 3D-space in fun and interesting ways resulting in a game of 'Where did it go now?'

Now that I know what to do, it actually makes perfect sense.  (Doesn't it usually?)

Clicking on one of the values under 'Part manipulation' displays a hint / arrow on the modifier but as soon as the mouse is moved over to the object, it disappears.  Can the corresponding move / scale / rotate mode in the main window be activated when the cursor is placed in one of the 'Part manipulation' fields?  

This would have given me the clue I needed when I was trying to learn it.

vojtechkral, 2019-07-23:
@rogerfroud This bugreport is also quite unintuitive. Could you please edit the title such that it's shorter and on point (just one line please). Please also edit the body of the issue - please fill in Slicer version (yes this is meaningful for us) and then describe the behaviour you would like to see in, preferably in a terse way if possible. To edit the issue, click the `•••` button in the top-right corner and then `Edit`.
Thank you.

rogerfroud, 2019-07-23:
vojtechkral - Ok, I'll change the title, but it would appear that this is intended behaviour, NOT a bug.

closed 2026-09-29 by prusaslicer-bot[bot]