prusa3d/PrusaSlicer #2661: Retraction should be a print setting

atomgomba, 2019-07-21:
I'm puzzled why retraction settings belong to the Printer settings group. Normally one would change those based on filament type, eg. more retraction in case of PETG, completely disable for flexible, etc. Hence I propose moving it to Filament settings. What's the reason for having to set these parameters along with the bed and nozzle size?

haarp, 2019-07-21:
It's probably a holdover from the days when all you'd ever print was PLA and ABS. I agree, retraction should be a filament setting.

While we are at it, extruder clearance should be a printer setting, not a print setting.

bubnikv, 2019-08-01:
retract settings could be overridden by filament profile in the upcoming PrusaSlicer 2.1.0-alpha1. Closing.

closed 2019-08-01 by bubnikv

alhafoudh, 2019-12-15:
Where can we find this feature?

lukasmatena, 2019-12-17:
Filament Settings->Filament Overrides

alhafoudh, 2019-12-17:
Ok, but can I do it per-object?

lukasmatena, 2019-12-17:
Not as far as I know. Is it necessary? As this issue title suggests, retraction should be a filament setting, shouldn't it?

atomgomba, 2019-12-17:
Please stop spamming, this is not a support forum.

alhafoudh, 2019-12-17:
@atomgomba um, sorry?

atomgomba, 2019-12-17:
@alhafoudh This is a place for discussion about software development and not about helping users. If you have questions about how to use the software, please go to Facebook or Reddit, etc. Or you can open a feature request too. This is getting totally unrelated to the original concern. If you want retraction setting per-object, why would it be a filament setting? On the other hand, the title of the issue is "Retraction should be a filament setting" and then you come asking where you can find this feature. Well, under Filament settings... Please stop spamming.

alhafoudh, 2019-12-17:
@atomgomba I have never encountered this kind of approach from community on GitHub or OpenSource. Maybe my question is not very verbose, point is if I am just missing feature which does not exist or I have overlooked existing feature.

I came to this issue from Google where I was trying to find how or if I can set retraction settings per-object. My opinion is that it is still valid question in this context.

I am not asking here how much prusa costs, I am discussing features of this software.