prusa3d/PrusaSlicer #2662: Firmware update fail

raxamus, 2019-07-21:
2.0.0+win64-201905201652

Windows 10 home

Prusa i3 mk3s firmware 3.7.2.1

Hey

I'm keeping getting this error: 

avrdude-slic3r -v -p atmega2560 -c wiring -P COM3 -b 115200 -D -U flash:w:0:C:\Users\raxam\Downloads\prusa3d_fw_3_7_2_MK3S\prusa3d_fw_MK3S_3_7_2_2363.hex:i

avrdude-slic3r: Version 6.3-20160220-prusa3d, compiled on May 20 2019 at 18:52:49
Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/ 
Copyright (c) 2007-2014 Joerg Wunsch

Using Port : COM3
Using Programmer : wiring
Overriding Baud Rate : 115200
avrdude-slic3r: stk500v2_recv(): timeout
avrdude-slic3r: stk500v2_getsync(): timeout communicating with programmer
avrdude-slic3r: Could not open port: COM3

avrdude-slic3r done. Thank you.

And i have tried, fac reset, reinstall drivers and slicer, tried another usb cabel, change com port, another computer and upload older firmware, but all make the same error.

Any ideas?

best regards

lars

bubnikv, 2019-07-22:
> avrdude-slic3r: Could not open port: COM3

Some other application is keeping the port open?
Do you have Cura or Repetier Host running?

raxamus, 2019-07-22:
Not what i could see when i tryed to change from com3 to com2, there did it state that com1 and com4 was the only one that was bust at the moment.
lars

bubnikv, 2019-08-09:
> Not what i could see when i tryed to change from com3 to com2, there did it state that com1 and com4 was the only one that was bust at the moment.

I don't quite understand. Are you saying, that the COM port identifier changed under your hands without you having to plug or unplog the device?

lsarios9144, 2019-08-26:
I keep getting the same error.

lsarios9144, 2019-08-26:
avrdude-slic3r -v -p atmega2560 -c wiring -P COM5 -b 115200 -D -U flash:w:0:C:\Users\luisa\Downloads\SWX1\Firmware - Main Board\Marlin 2.0 BL Touch.hex:i

avrdude-slic3r: Version 6.3-20160220-prusa3d, compiled on May 20 2019 at 18:52:49
                Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
                Copyright (c) 2007-2014 Joerg Wunsch

                Using Port                    : COM5
                Using Programmer              : wiring
                Overriding Baud Rate          : 115200
avrdude-slic3r: ser_open(): can't set com-state for "\\.\COM5"
avrdude-slic3r: Could not open port: COM5

avrdude-slic3r done.  Thank you.

bubnikv, 2019-08-26:
Any application is holding the serial port open? Cura?

lsarios9144, 2019-08-26:
No, I shut down all other programs and even rebooted the computer several times. I am not sure what is going on but I sure wish someone would figure it out. I really need to update the firmware.

bubnikv, 2019-08-27:
@lsarios9144 what message are you getting exactly?

lsarios9144, 2019-08-27:
bubnikv

> avrdude-slic3r -v -p atmega2560 -c wiring -P COM5 -b 115200 -D -U
 > flash:w:0:C:\Users\luisa\Downloads\SWX1\Firmware - Main Board\Marlin 2.0 BL
 > Touch.hex:i
 >
 > avrdude-slic3r: Version 6.3-20160220-prusa3d, compiled on May 20 2019 at
 > 18:52:49
 > Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
 > Copyright (c) 2007-2014 Joerg Wunsch
 >
 > Using Port : COM5
 > Using Programmer : wiring
 > Overriding Baud Rate : 115200
 >
 > avrdude-slic3r: ser_open(): can't set com-state for "\.\COM5"
 > avrdude-slic3r: Could not open port: COM5
 >
 > avrdude-slic3r done. Thank you.

bubnikv, 2019-08-28:
> @lsarios9144 <https://github.com/lsarios9144> Because of how github
> mangles slashes sometimes, is that supposed to be
> avrdude-slic3r: ser_open(): can't set com-state for "\\.\COM5" ?
>
> If not that is odd, WMI namespace should read \\.\COM5 which would be
> match how the vanilla version of avrdude logs look like.

Thanks for heads up. The following lenghty discussion may help us.
https://answers.microsoft.com/en-us/insider/forum/insider_wintp-insider_devices/windows-10-serial-usb-problems/438de66f-7294-4c06-b4fb-89b45d005ca0

bubnikv, 2019-08-28:
By the way, do you guys have your Windows 10 up to date?

lsarios9144, 2019-08-28:
Yes my Windows 10 is up to date. I found the solution. The steps are a little outdated but this method worked for me.

https://www.youtube.com/watch?v=8Y_78eDEQyY&t=190s

billy1380, 2020-06-05:
I had the same issue when Cura was running! It is obvious in hind sight, but just in case someone else is wondering. It is basically contention on the com port.

transitmark, 2020-07-19:
Thanks @billy1380!  This was my problem!

rtyr, 2021-02-03:
Likely resolved. Closing.

closed 2021-02-03 by rtyr

tomvocas6, 2022-07-03:
Hallo everyone. Same problem, but after change on Prusa3D printer ( at other printers setting ) com4 to com9 it is o.k. You may have sitting other soft. on com port.