prusa3d/PrusaSlicer #2663: Unrecognized Repetier Custom M codes

walterbux, 2019-07-21:
### Version
_Version PrusaSlicer-2.0.0+win64_

### Operating system type + version
Windows 10 Pro Version 1903

### 3D printer brand / version + firmware version (if known)
_Custom Built with Repetier FW v.1.0.3_

### Behavior
* _Describe the problem_

If inside the Tool Change Custom G-Code I use the following commands:
M401 - Store x, y and z position
...
G1 Z0
...
M402 - Go to stored position. (so restoring the Z of the last printed layer)

the  3d preview in the layer section of the plater becomes corrupted at the end of the processing by squeezing the tool change layers to Z=0 level.

* _Steps needed to reproduce the problem_
  load a set of stl files as a multi material object or an amf/3mf file
assign colours to each part of the object
process and export gcode to a file.
NOTE: wipe towers are disabled !!
* _Expected Results_
3d preview of the object not corrupted both in prusaslicer and inside repetier host application

* _Actual Results_
  the 3d preview of the object is corrupted both in prusaslicer and inside repetier host application
NOTE: If I enable the wipe towers the problem disappears
_Is this a new feature request?_
NO
#### Project File (.3MF) where problem occurs
Any multi material object.

enricoturri1966, 2019-07-22:
Working on it -> https://dev.prusa3d.com/browse/SPE-990

enricoturri1966, 2019-07-22:
@walterbux
Could you please attach here the 3mf file you are using ?
Despite I can confirm that PrusaSlicer is not handling the gcode lines M401 and M402, I am unable to repro your issue

walterbux, 2019-07-24:
Sure, here it is.

Walter
[diamond test.zip](https://github.com/prusa3d/PrusaSlicer/files/3425283/diamond.test.zip)

FidelCapo, 2019-07-25:
Thank you for report. This issue should be fixed in next release.
Closing.

walterbux, 2019-09-01:
When is planned a fix for this issue ?
Thanks.

Walter

bubnikv, 2019-09-02:
It should be fixed in the 2.1.0-beta3 already.

walterbux, 2019-09-02:
Unfortunately the problem still exist in the  2.1.0-beta3. I have just
repeated the test with the file already provided.
I wait for news.
Thanks.

Walter

FidelCapo, 2019-09-02:
I think changing Printer settings - Firmware - G-code flavor from RepRap/Sprinter to Repetier in your project will solve this issue.

walterbux, 2019-09-02:
Yes you are rigth. Just tested.
Thank you.

Walter

walterbux, 2019-09-28:
I am sorry to inform that in the PrusaSlicer-2.1.0+win64-201909160915 there are still problems in the management of the Repetier M codes:
If inside the Tool Change Custom G-Code I use the following commands:
M401 - Store x, y and z position
...
G1 Z0
G1 Z5
...
M402 - Go to stored position.

for more than 3 times inside the first layer I obtain that from the 4th switch on the M402 - Go to stored position goes to the wrong layer.

In practice I am printing 6 cubes of different colors with 6 different extruders.

When I change color I prime the next extruder in a side area of the hotbed.

From the 4th to the 6th time that i do this when I come back to the first layer printing with the M402 It seems that I am printing the upper layer because the extruder push out material flying on the hotbed at about 1 mm height.
If I repeat the same print without these Tool Change Custom G-Codes and everything works perfectly.
Many thanks for your help.

Walter

lukasmatena, 2019-09-28:
Ok, reopening the issue.

I'm sorry but I cannot make any sense of it. Please, provide a 3MF with a minimal example. Also, an excerpt from the output gcode showing what exactly is wrong would be appreciated.

walterbux, 2019-09-28:
[Mcodes test prusaslicer.zip](https://github.com/prusa3d/PrusaSlicer/files/3665628/Mcodes.test.prusaslicer.zip)
The purpose of these files is to test the use of 2 diamond extruders working toghether but only 6 colors are printed: the colors of the PLA wires. In the test no mixing is done.
The file "Test-2-diamond-6-colori" include the Mcodes that are critical and exhibit the wrong behaviour whilst the file "Test-2-diamond-6-colori-noclean" works properly.
I have not been able to find strange things in the Gcode at the switching of the T16 extruder line 870...
Thank you.
Walter

lukasmatena, 2019-09-29:
I don't see how the first three toolchanges differ from the ones that cause issues. If you yourself see nothing wrong in the gcode, it could quite well be caused by the printer interpreting the gcode incorrectly.

I should say I have no experience with diamond hotends and am no expert on Repetier firmware either. Can anyone see anything wrong with what PrusaSlicer generates here?

walterbux, 2019-09-30:
Hi Lukas,
you confirm what I was suspecting. I don't see anything wrong in the gcode.
I am doing experiments with the Diamond Hotend and I use a Repetier Firmware very lightly modified to manage 32 virtual extruders instead of 16 max of the standard Repetier Firmware.
I have modified the Firmware strictly following the indications received by the Repetier Forum.
I will report the issue extending the first request to Repetier Forum.
Many Thanks for your useful work.
I will probably make a pull request to PrusaSlicer forum to better manage this type of extruder at the end of my experiments.
Regards.

Walter

lukasmatena, 2019-09-30:
Ok, if there is nothing wrong with the gcode, I'll reclose the issue then. It's about handling M401/M402 commands and that seems to be working fine.

closed 2019-09-30 by lukasmatena