prusa3d/PrusaSlicer #2664: Fan is not disabled in gcode

akhonkey, 2019-07-21:
### Version
2.0.0+win64

### Operating system type + version
Windows 10 Pro build 1809

### 3D printer brand / version + firmware version (if known)
Prusa MK3S MMU2S, firmware: 3_7_2 MK3S ; 1_0_5 MMU2S 

### Behavior
When choosing Flexible filament, the fan is not disabled for the entire print job even though the filament settings are configured to do so.
[image]

Upon inspecting gcode, the fan turns on full speed after the first layer

#### Project File (.3MF) where problem occurs

[FLEX_MK3SMMU2S.zip](https://github.com/prusa3d/PrusaSlicer/files/3415152/FLEX_MK3SMMU2S.zip)

bubnikv, 2019-07-22:
The fan is still enabled for the bridges, which is indicated by the UI by
not graying out the "Bridges fan speed" field.

akhonkey, 2019-07-23:
Should it just enable the fan for layers that have bridges then? The objects I was printing do not have any bridging. My solution is to just remove all the gcode commands turning the fan on, just a bit annoying to do that. 

Jon

bubnikv, 2019-07-29:
The first dense infill layer over sparse infill is printed with a bridging
flow.

akhonkey, 2019-07-29:
Ok. So to entirely disable the fan I need to set the bridging fan speed to 0...I’ll give that a try and let you know how it works out

Jon

rtyr, 2020-05-25:
Explained, I believe we can close this.

closed 2020-05-25 by rtyr