prusa3d/PrusaSlicer #2666: Prusa Slicer V2.0.0 thin wall dome print causes holes from infill to perimeter transition

TimothyGold, 2019-07-22:
### Version
PU Slic3r 2.0 

### Operating system type + version
Win64

### 3D printer brand / version + firmware version (if known)
MK3S MMU2S

### Behavior
STL - GCODE transition from infill to perimeter in thin wall starts to chatter infill and leave holes at layer #357

#### Project File (.3MF) where problem occurs
[RunCamSplit Cover.zip](https://github.com/prusa3d/PrusaSlicer/files/3415924/RunCamSplit.Cover.zip)

neophyl, 2019-07-22:
Looks like the change from gap infill to internal perimeter is causing the algorithm some issues, you can see the arc of the curve for the dome in the small traces it is trying to lay down.  There isnt room for a full perimeter as the model isn't actually a smooth curve but broken into a series of straight lines, its at these slight 'corners' where the lines join that its deciding that its not wide enough for a perimeter line  so its stopping.  In effect you get a dotted line around the object.

If you enable the checkbox for detect thin walls it seems to slice without the effect and looks a lot better.
[image]

give that a go and see if it works better.

TimothyGold, 2019-07-22:
I did actually try enabling "detect thin walls" and it didn't seem to look any better.  Even in the preview it looked about the same as without it checked.  I didn't actually print with it enabled but will try next go around.  I just didn't have a lot of confidence that it would be different in the print if the preview didn't look any different.

TimothyGold, 2019-07-22:
[image]
Here is a preview shot of with "Detect thin Walls" enabled.  it still seems to show the gaps.

neophyl, 2019-07-23:
Yeah sorry I didnt notice that they problem just moved to a different part of the domes wall structure.
I couldnt find any combination of settings in Prusa Slicer that would slice it without that short perimeter problem.

However there is a chap named Supermerill that has a fork built upon the latest PS build that has a few other settings and features in at https://github.com/supermerill/Slic3r/releases  (it also has some of the fixes/changes that PS next release will have just a bit early)

There is a setting in that build called Curve Smoothing under Print Settings>Slicing that when I set it to 5 degrees angle does appear to slice it fine from what I can see.  It might be worth downloading it and giving it a try.  You don't need to install it as it will run from the downloaded folder and it can be run in parallel with the official PS release (just export your configs as a bundle and then import them into the ++ version.  Worth a shot anyway.

TimothyGold, 2019-07-23:
Thanks again for your help.  I certainly will give it a try....

Jebtrix, 2019-07-24:
Depending on the model I've been able to use slab modifier to tweak infill/perimeter width on problematic layers. Here is your model "fixed".
[image]
[image]

[RunCamSplit Cover-Tweaked.zip](https://github.com/prusa3d/PrusaSlicer/files/3428643/RunCamSplit.Cover-Tweaked.zip)

TimothyGold, 2019-07-24:
Thanks for the information.  I did take your slab modifier idea and applied it to a newer revision of the part with some even smaller Perimeter Width set to .3 and it seems to have fixed the transition.  The features to get this done are awesome.  There are still some definite bugs in trying to scaled and move a slab into position.  By default the slab is placed in the center of the part and if you resize the Z and then move it down or up it is referenced from the center of the part and not from the print base.  This makes it very hard to accurately place a part at a specific measured layer height which is found from the sliced model.  Overall it keeps getting better and better.  Hopefully the devs can replicate this issue and refine the placement of modifiers.  

Thanks again..

Jebtrix, 2019-07-24:
Yes, the placement mechanics of modifiers is its _painful_ achilles heel.  Glad you were able to apply the idea successfully. Would you believe I eyeballed placing that slab on the first try? Pure luck :) I have worked around challenging slice issues countless times using generic and custom object modifiers. Once you get used to using them PrusaSlicer really shines.

closed 2026-09-29 by prusaslicer-bot[bot]