prusa3d/PrusaSlicer #2669: Editor Screen Icons are blued out

ChasBow77, 2019-07-23:
### Version
_PrusaSlicer 2.0.0 + win64_
[image]

### Operating system type + version
_Windows 10 Home \ Version 1903_
[image]

### 3D printer brand / version + firmware version (if known)
_Prusa I3 MK3 \ Frimware 3.7.1-2266_

### Behavior
* _Describe the problem_
 _**All of the Icons on the Editor Screen are blued out. They do work but I cannot see what is what.**_
[image]

_**There is also a second window that opens when I run Prusa Slicer**_
[image]

_Is this a new feature request? No_

bubnikv, 2019-07-29:
It seems like your system OpenGL driver does not support OpenGL 2.0, the
bundled MESA OpenGL driver was loaded, but the system OpenGL driver was not
uploaded correctly.
You may try to copy the  mesa\opengl32.dll one directory level up (to the
main slicer directory).

bubnikv, 2019-08-23:
PrusaSlicer supports OpenGL 2.0 and newer. The upcoming PrusaSlicer 2.1.0-beta will provide an appropriate error message with an advice how to execute PrusaSlicer with software rendering.
Closing.

closed 2019-08-23 by bubnikv