prusa3d/PrusaSlicer #2670: Wipe tower missing first few layers if printing floating object on supprts

Rassah, 2019-07-23:
### Version
2.0.0+Win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa MK3s + MMU2S

### Behavior
This problem was actually first noticed in previous versions, and seems to have been carried over into the current version. If you print a multi-color object with pointy ends down, so it has to float and be printed on top of supports (think upside-down cone), Slic3r will print support base first, then the first lines over the support, and then print the tower starting at the second layer height. It ends up missing the brim part, and not sticking to the bed. There needs to be some way to enforce the tower to start printing first with the brim and first layer.

lukasmatena, 2019-08-08:
I believe this error should be fixed with https://github.com/prusa3d/PrusaSlicer/commit/6ab1cec48c037a4f4a132a06e83f2a12fcacd9ff. The fix will be part of upcoming PrusaSlicer-2.1.0-alpha1, which will be released in a few days. Could you please retest in that version and close the issue if happy? Thanks.

lukasmatena, 2019-10-09:
I just retested with following 3MF (pyramid with downward-pointing tip). PrusaSlicer 2.1.0 slices it fine, PrusaSlicer 2.0.0 crashes. I believe the issue is indeed gone.
[pyramid1.zip](https://github.com/prusa3d/PrusaSlicer/files/3706059/pyramid1.zip)

Closing the issue.

closed 2019-10-09 by lukasmatena