prusa3d/PrusaSlicer #2672: Color change returns to print in a location where new color is not needed/wanted?

stevenminnick, 2019-07-23:
### Version
2.0

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa Mk3s

### Behavior
I posted this over here:
https://github.com/prusa3d/Prusa-Firmware/issues/2033
But was told it belongs here.

I've noticed some weird behavior and wasn't sure if this is on purpose.  

This is what I want to print
[image]

All pretty

But when the first color finishes it finishes at a corner of the print, lets assume where I draw the arrow.
[image]

I do the color change. And the print head moves back to the print, but instead of going where the new color begins it goes back to where the old color left off. So it deposits some new color here, and then moves to where the new color starts.
[image]

So the print comes out like this. I can remove that little dot with tweezers. It's not that big of a problem.
[image]

Just seemed like an oversight. Or a bug. Why would it go right to the previous end point, deposit color, then move to where the new color is supposed to start.

Maybe I didn't notice it on previous firmware, but I've been noticing it lately once I swapped over to Prusa Slicer 2.0 (now that color change is built in) and the latest couple of firmware releases.

_Is this a new feature request?_
Yes?  Maybe?

Maybe it's just the placement of that M600 command.  Not sure if there is a better way to do it.  

If no one thinks it's an issue, don't worry about it.  It just seemed new and not expected so I figured I would mention it, maybe I'm the only one who notices.

stevenminnick, 2019-07-23:
Maybe this is more of a feature request, but I wonder if it's worth moving the print head to the new start location on the new layer, then change colors.  Then move back.  

Just thinking out loud.

Panayiotis-git, 2019-07-31:
This happens if you add the M600 command at the tool change custom g-code field. At this point you don't know where the next location will be.

stevenminnick, 2019-07-31:
I don't 100% understand how it works, but my understanding is that the M600 command is "dumb" and correct it doesn't "know" where the next location will be.

1. it doesn't have to be that way.
2. you could in the slicer figure out where the next location will be, put the print head there, then M600 and then go back to the next location.

The system as a whole knows where the next location will be.

But maybe no one else thinks it's an issue or even needed.

bubnikv, 2021-02-15:
I don't think this is a big deal, but it would be better solved in the firmware. Namely, the firmware should skip all the travel lines after M600. There may be a devel in the detail, for example how to handle possible retracts following M600, but I think it is doable.

bubnikv, 2021-02-15:
Namely, at least for Prusa's Marlin fork:

Slicer does not know the XY coordinate of where the filament exchange is being performed.
It is the firmware, that does the travel to the filament exchange position and which returns to the position before M600. There is nothing we can do in slicer about that. It has to be handled by the firmware.

espr14, 2021-02-15:
PrusaSlicer 2.3 G code:
```
;LAYER_CHANGE
;Z:1.2
;HEIGHT:0.2
;BEFORE_LAYER_CHANGE
G92 E0.0
;1.2

G1 Z1.200 F59940.000
;AFTER_LAYER_CHANGE
;1.2
;COLOR_CHANGE,T0
M600
G1 E-3.20000 F4200.000
G1 Z1.400 F59940.000
G1 X77.938 Y88.132
G1 Z1.200
G1 E3.20000 F2400.000
M204 S500
;TYPE:Perimeter
;WIDTH:0.45
G1 F900
G1 X77.938 Y91.868 E0.12645
```
Current output in words:
1. finish layer
2. **change filament**
3. retract
4. raise head
5. move to next layer (XY)
6. lower head
7. **deretract**
8. print next layer

Proposed order:
1. finish layer
3. retract
4. raise head
5. move to next layer (XY)
7. **deretract**
2. **change filament**
6. lower head
8. print next layer

FW could merge retraction and deratraction in G code with retraction and deretraction in M600. But it's not a good idea to analyze G code further and merge other motions. For example this could destroy "Avoid crossing perimeters" path. Moreover, there are very limited resources (flash, RAM, flops). It's much easier to add M600 parking place in the Printer Settings in Prusa Slicer and plan the parking in Prusa Slicer. That allows avoiding crossing perimeters during parking which is not possible in FW.

stevenminnick, 2021-02-15:
> Current output in words:
> 
>     1. finish layer
> 
>     2. **change filament**
> 
>     3. retract
> 
>     4. raise head
> 
>     5. move to next layer (XY)
> 
>     6. lower head
> 
>     7. **deretract**
> 
>     8. print next layer
> 
> 
> Proposed order:
> 
>     1. finish layer
> 
>     2. retract
> 
>     3. raise head
> 
>     4. move to next layer (XY)
> 
>     5. **deretract**
> 
>     6. **change filament**
> 
>     7. lower head
> 
>     8. print next layer

I didn't think anyone still looked at this.  :-)

But yes, what you put here is what I was originally thinking of.  It's not a FW thing, it would be a slicer thing.  You put it better than I did.  If the print head went to the next layer first, then did the change filament, and if the new color wasn't fully cleaned off the nozzle when it was done with the color change it would go back to the x/y location where it was going to start extruding anyway.  

It's not a huge deal, maybe I'm the only one who experiences this.  Maybe it's not a big deal to make where the change filament code is placed in the g-code as a setting (at the end of the last layer or the beginning of the new layer).

cQszasza, 2021-08-17:
Finally I've found someone who uses their eyes. How this bug does not bother anyone else? Why would I need to remove the little blob after the print finishes with a pair of tweezers or something when it could be avoided by software?
I'm doing a lot of mid print colour changes and this thing bugs the hell out of me. Imagine 2 very different colours, sometimes this blob just melts in too deeply so its not possible to remove it without obvious scratch marks on the print.
Devs, please read OP's post carefully because it seems like some of you don't properly understand what the problem is. It would be so nice to have this sorted, let me know if I can help in any way, maybe even with just photos of test prints or anything really. Thank you!

ssill2, 2021-08-17:
+1 for this.  I print a lot of keychains in tpu that have a single color change.  The color change ALWAYS does what you're describing and usually I try to catch the filament with tweezers so that little or none drops on the other color.  It would be great if the slicer could be made to start where print is actually going to put down filament in the new color.

bdwheele, 2021-09-05:
I reported this issue this afternoon (which got closed as a duplicate) and somehow didn't find this ticket before I reported it.  Yes, this is something that bothers other people!

jeremyreeder, 2022-02-11:
I'm not very familiar with PrusaSlicer's source code, but I believe I've located the problem in libslic3r/GCode.cpp and that it could be fixed by making the change shown here (which I have not tested):
https://github.com/jeremyreeder/PrusaSlicer/commit/61b261ccfd67a93f031b7d4b5112f3418dc95387

lukasmatena, 2022-02-15:
This issue will probably be addressed soon, we have discussed this internally.

@jeremyreeder 
> which I have not tested

Why not?

jeremyreeder, 2022-02-15:
Because I do not yet know how to build PrusaSlicer from source. I'm working on that now.

Nohus, 2022-10-12:
@lukasmatena 
> This issue will probably be addressed soon, we have discussed this internally.

Is this still in progress? Any way I could help?

Nohus, 2022-10-13:
I went ahead and fixed it in #9036

RAMilewski, 2023-01-24:
Everyone here please go to #9036 and add a +1 to merge ASAP.

stevenminnick, 2023-01-24:
> Everyone here please go to #9036 and add a +1 to merge ASAP.

I apologize for my lack of github etiquette knowledge, but what do you mean by "add a +1 to merge"?  Is that adding just a comment?  Or is there something else in the system that would float this fix higher in the system?

MattBlissett, 2023-02-19:
A quick workaround is to add something like this before the `M600` in the  tool change G-code:

```
G1 Z{layer_z+2} F720; go up a bit more
G1 X210 Y-3 F10800; go outside print area
```

The `layer_z` variable isn't available in the colour change G-code, but the second line can still be used.

Since this moves the extruder to the edge of the bed, near where the firmware moves it for the colour change, after the colour change the firmware returns in to this position. The next move command is to the start of the object with the new colour, avoiding the dot.

> what do you mean by "add a +1 to merge"?

That means pressing the smiley face button on the first comment and giving it a thumbs up or similar. It's better than commenting "+1 great" as it avoids irrelevant email notifications to the developers.

takoeninger, 2023-04-20:
Same problem for me, but a bit more serious.  No matter how I position the part, the last bit to be printed before color change (which will bridge the opening) is the tiny dot of material in the numeral "6".  After color change, the nozzle touches down on that dot & pulls it off.
[image]

RAMilewski, 2023-04-23:
While we'e all waiting for a fix in Prusa Slicer, check out:

https://github.com/awabom/3dprint/tree/main/PrusaSlicer-FixM600

It's a script for you can call in "Print Settngs > Output Options > Post-processing scripts" that reworks the gcode and flattens the problem. Originally written for, and tested on, an Ender-3, it works fine on my Prusa Mini (and, I suspect, any FDN printer), I've used it on Prusa Slicer 2.5 and the 2.6 Alpha 6 (the most current at this writing).

takoeninger, 2023-04-23:
Thanks for the info, it is really helpful.  I've been using a "helper disc"
in PrusaSlicer to "lure" the nozzle away at the end of the layer instead of
the dot in the numeral 6, so this will be a welcome change.

ssill2, 2023-09-07:
this still happens in 2.6.1.   Please prusa fix this.  Currently printing some tpu keychains and this issue happens twice on each print.

espr14, 2023-09-24:
Another alternative for **Prusa MINI/MK4/XL** is using `M600 N` which does not return to the original location. How to setup:
1. Go to `Printer Settings` tab
2. Choose `Custom G-code` on the left
3. Enter `M600 N` to `Color Change G-code` box

[image]

BrienKing, 2023-10-03:
The M600 N did not work for me.

espr14, 2023-10-06:
> The M600 N did not work for me on the Prusa XL

The code is there for all Prusa printers. The issue is the generated G code. It should not return back to the exactly the same spot but it can return to the next path segment. Also the Z should be lifted already so the damage should lower.

isokoxyz, 2024-02-05:
this is still happening to me on my new mk4 and mk3s+ ...... ples someone help

isokoxyz, 2024-02-05:
the "M600 N" works fine for mk4 but not on mk3s+, i see it was marked closed but still facing this issue, any ideas?

Nohus, 2024-02-05:
> the "M600 N" works fine for mk4 but not on mk3s+, i see it was marked closed but still facing this issue, any ideas?

This problem was fixed in PrusaSlicer in my PR: https://github.com/prusa3d/PrusaSlicer/pull/9036
But the version of PrusaSlicer containing the change is not released yet.

closed 2024-02-09 by lukasmatena

jordanbrown0, 2024-12-28:
I encountered exactly the opposite problem:  just before the pause, it moves to the first location for the new color, drops a blob of the **old** color, then the pause drags a string of old color across the model.  For a name tag this is pretty obvious; the first-new-color location tends to be in the middle of the big flat area.  I don't think I can directly fix it with custom G-code, because it looks like it:
* Jumps Z to the next layer
* Retracts
* Moves XY to the last position in the old color (where it already is)
* Moves XY to the first position in the old color
* **Un-retracts**
* Runs the custom color-change gcode
* Draws the first stroke of the new color
So it's already dropped the blob by the time it pauses.

I might be able to make it better by having the custom color change retract before pausing, and un-retract after pausing.  That would presumably help with trailing string, and might suck back in some of the blob.  (Perhaps the printer firmware - CraftBot+ - should do that, but it looks like it doesn't, and it doesn't know how much is a good amount to retract.)

It seems like it would be safest to retract, move to the new location, pause, then un-retract.  Maybe with a Z-hop or two in there.

jordanbrown0, 2024-12-28:
That's with PrusaSlicer 2.9.0 on a CraftBot+.

jordanbrown0, 2024-12-28:
Actually, no un-retract needed after the pause; presumably the user has fully loaded the nozzle before resuming.