prusa3d/PrusaSlicer #2673: Not generating support material

mineshaftgap, 2019-07-23:
### Version
2.0.0+

### Operating system type + version
macOS 10.14.5 (18F132)

### 3D printer brand / version + firmware version (if known)
ORIGINAL PRUSA I3 MK3S + 3.7.1

### Behavior
* Not generating support automatically or manually
* _Steps_
  * Load STL file
  * Printer Settings -> Advance
  * Check "Generate Support Material"
  * Use default "Overhang Thresh" or set to 0.
  * Platter -> Supports -> Everywhere

* _Expected Results_
  * Generate support material for overhangs

* _Actual Results_
  * No support material generated
[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[assets.zip](https://github.com/prusa3d/PrusaSlicer/files/3423018/assets.zip)

mineshaftgap, 2019-07-23:
It's worth noting that I have also tried to add support material to no avail via an enforcer.

neophyl, 2019-07-23:
Can’t open the file on my iPad but looking at your image the light blue means slicer is treating it as bridging layers so the model doesn’t need any support. It may actually be right but I think you will get a bit of sag. Might be worth trying though. 
If you really want support then turn off bridging.

mineshaftgap, 2019-07-23:
@neophyl thanks, I will try that tonight.

mineshaftgap, 2019-07-23:
@neophyl as unchecked "Don't support bridges" I got support. That is a default.

jwillmer, 2019-07-29:
@mineshaftgap can be closed?

mineshaftgap, 2019-07-30:
@jwillmer yes, thanks

closed 2019-07-30 by mineshaftgap