prusa3d/PrusaSlicer #2675: Feature Request: Full height skirt checkbox (thermal shock cover for ASA)

mikolaszuza, 2019-07-24:
Add a checkbox for a full-height skirt, preferably next to or above the current skirt height setting. Ticking it would grey out (inactive) the height input and would behave just as if the height is set to a really big number, which is the current workaround.

**Why would you want to print full height skirt?**
It works as a thermal shock cover for ASA/ABS prints, they prevent parts from cracking and warping and allow even relatively big objects to be printed without an enclosure.

**Why would you want a checkbox for this?**
So that it's discoverable. When you look at skirt settings, you will not realize that full height skirt is something to consider. When there is a dedicated checkbox to it with a tooltip - e.g. _"Prevents drafts and keeps a constant temperature around the printed part, which decreases warping and layer separation. Useful especially for ASA and ABS."_ user can find such setting and consider using it. Also, it's a nice quality of life thing, inputting 9999 as height becomes annoying really quickly.

[image]

[image]

The amount of wasted filament for such skirt is very small compared to the overall filament use and compared to wasted filament on a failed print due to warping.

wavexx, 2020-02-22:
Skirt height can now be specified in the "Filament Settings -> Filament override" options, so that you can set ABS to always have a full skirt (err.. large number ;)).

This behavior is now very acceptable to me

wavexx, 2020-02-22:
Woops, looks like I was fooled by a mash-up screenshot :/
This is the feature request that would do it for me: #3558

bubnikv, 2020-03-14:
A "draft shield" option will be imlemented in PrusaSlicer 2.2.0-rc4.
Closing.

closed 2020-03-14 by bubnikv