prusa3d/PrusaSlicer #2678: Slicer doesnt recognize multi part STL in single material mode

smejfi, 2019-07-25:
### Version
2.0.0

### Operating system type + version
Windows 10 Home
### 3D printer brand / version + firmware version (if known)
Prusa MK3 3.7.2

### Behavior
If I want to import multi-part STL files into prusa slicer and the printer is set to anything else than MMU2S, it doesnt recognise that it is multi-part STL and imports it as separate parts. One can ask why would I want to import multi-part model when I dont want to print multi material, but I find it handy that it is possible to change some print settings separately for each part of the model (eg. extrusion width etc.). Now this is possible only when I set printer as MMU2S, then I import multi-part STL and then I change the printer back to MK3S for example. It is nothing serious, but why not fix it...

Jebtrix, 2019-07-25:
Well the initial import behavior is different but you can right click on the object and select "Split > To Parts" to get what you want. I would not use "Split > To Objects" because that breaks the Reload from Disk functionality.

bubnikv, 2019-08-01:
@smejfi does the explanation given by @Jebtrix satisfy your needs?

smejfi, 2019-08-02:
I already knew this part, but that is not handy in case that my model consists of a lot of parts because slicer would separate all of them. I meant the case where I have multipart object already prepared in meshmixer and I just want to import stl files such that their mutual positions will remain the same as in meshmixer. But I have already figured it out using the function "Add part", or, as I have already mentioned, it is also possible to use MMU2S profile and then switch to the clasic MK3S profile, so, yes this topic can be closed from my point of view, thank you for asking and thank @Jebtrix  for answer.

bubnikv, 2019-08-09:
thanks, closing.

closed 2019-08-09 by bubnikv