prusa3d/PrusaSlicer #2679: [No Issue] Feedback on successful compilation with VSCode 16 2019

FHeilmann, 2019-07-25:
I just wanted to comment that I was able to successfully compile PrusaSlicer 2.0.0 (commit  1a3fc09) in Visual Studio 16 2019. The build instructions apply 1 to 1, except of course that you need to compile the dependencies yourself.

For those that want to use Visual Studio 16 2019 I have attached zip files with the dependencies compiled in release and debug+release mode. I hope this helps others.

[PrusaSlicer deps win64 RELEASE](https://drive.google.com/file/d/1OR78l2VHdiyJ46mJcTO52lZ3pieJJhan/view?usp=sharing)

[PrusaSlicer deps win64 RELEASE+DEBUG](https://drive.google.com/file/d/14bqn5HDfGQWLRUy8Rxsu0V1GfDM8Zq7x/view?usp=sharing)

Note that VS2019 only supports 64 bit builds.

Bests,
-Florian

bubnikv, 2019-07-26:
> Note that VS2019 only supports 64 bit builds.

This is not correct. We compile our 32 bit Windows builds with VS2019

FHeilmann, 2019-07-26:
My bad. Msbuild didn't take the arch argument on my machine so I assumed it was 64bit only. I'll adjust my original post

closed 2019-07-26 by FHeilmann