prusa3d/PrusaSlicer #2680: Request: controlling inside brim

neslekkim, 2019-07-26:
### Version
2.0

[image]

When printing some thing, like these clips, (https://www.thingiverse.com/thing:3757356) and other small items, I need to use brim when printing ASA.
It would have been nice, if I could use an modifier thingy, to mark where there should be an virtual perimeter, so I don't get brim inside the objekt like on this picture.
Brim inside this is of not much use, and I do know i could have designed the object to have an tiny part to cut away after printing, but when printing objects by others, it's not always so easy to do.

I have also tried to add things like https://www.thingiverse.com/thing:217725 helper disc, corners helpers etc, but the problem is that PrusaSlicer does not regard aditional objects placed inside eachother when creating the brim, so I cannot use an external object to sort of close the perimetre in the slicer.

tkircher, 2019-07-27:
Just create a one-layer-high brim of the correct shape in your 3D modeling program. It's simple, and the developers have made it clear that brim and support structure issues are a low priority.

neslekkim, 2019-07-27:
As I said, I know I could have done it in design.

But, if the slicer could combine objects that are on top of each other, BEFORE it creates the brim, everything would have been solved.

And yes, I agree that these things are low priority, but does that mean they cannot be asked? :)

Jebtrix, 2019-08-12:
Technically you can control brim with "modifiers", though its **Add Part** to object to do it. It takes some practice to figure out how to place them properly on Z but you can get pretty fast at it. The main rule is added object primitive has to be first layer height in height. Once you place one you just copy and place as many as it takes, (turn off Scale Lock to not mess with height) as you scale the copies to fit. Once you have one then you can duplicate the object which will duplicate all the object mods with it.

[UniversalDualFilamentDragonClip-thick-ManualBrimControl.zip](https://github.com/prusa3d/PrusaSlicer/files/3493839/UniversalDualFilamentDragonClip-thick-ManualBrimControl.zip)

[image]
[image]

neslekkim, 2019-08-12:
Great!, I saw this add part thing when I installed the alpha now, and wondered if it was a new thing, I guess not. 
I need to try this, this is what I thought should be possible when I added an extra object, but adding it that way might be the trick.
Thanks!, I need to work on this.

pacaj2am, 2019-10-19:
I'd like to be able to turn of brim completely for some parts on the build plate. Right now it is difficult to remove brim from some parts.

neclimdul, 2026-09-18:
I'm not sure it makes sense to start closing issues because an incomplete alpha is out. An alpha that's according to the release not fully featured for Prusa printers and doesn't support even support non-prusa printers so a large user base can't even test and confirm issues like this.