prusa3d/PrusaSlicer #2681: Filament settings are bugged

Schild0r, 2019-07-27:
### Version
V2.0.0+win64

### Operating system type + version
MS Windows 10 Pro
Version 10.0.17763 Build 17763

### 3D printer brand / version + firmware version (if known)
Custom printer with Marlin 1.1.9 but thats besides the point

### Behavior
Sometimes not all of my Filament Settings are shown.
I installed PrusaSlicer and manually transferred all the settings. After opening it again some of my presets were not shown but as you can see they are still there technically in the folder
I don't know why or when this happened but somehow it fixes itself after some days or so but then the settings go away again.

Sorry for the "screenshot" but I couldnt take a screenshot with the dropdown open
[image]

Schild0r, 2019-07-27:
Okay so obviously when I edit the filament settings ini file (in notepad) and remove the dependency it will work

bubnikv, 2019-07-29:
So is your issue resolved? Could you close this issue?

Schild0r, 2019-07-29:
Well actually no since this is only a workaround.
I think this would be a problem for any profile dependency.
I could reproduce it by setting a dependency and then deleting the profile which the dependency refers:
The profile which had the dependency to a not anymore existing profile would not be shown.

bubnikv, 2019-07-29:
>  The profile which had the dependency to a not anymore existing profile
would not be shown.

Why is this Slicer's bug? We consider it to be an expected behavior, if you
for example uninstall a printer, with which the filament profile is not
compatible.

Schild0r, 2019-07-29:
Well this is what the flags are for right?
Maybe you want to build upon your "old" profiles instead of starting from scratch.
I can understand your point but then a hint would be nice like "you are deleting a profile other profiles are dependent on. These profiles will not be visible anymore too. Are you sure?"

Also I am not sure whether this (deleted profile) was the case in the first place when I had this problem. Just a way to recreate it

bubnikv, 2019-07-30:
There is a checkbox in the application preferences to allow you to show all
the profiles, even the incompatible ones. If you enable that, you will get
a flag at the top bar of the preset page, which toggles between red
(non-filtered) and green (filtered).

Schild0r, 2019-07-30:
> There is a checkbox in the application preferences to allow you to show all
the profiles, even the incompatible ones. If you enable that, you will get
a flag at the top bar of the preset page, which toggles between red
(non-filtered) and green (filtered).

thanks I guess that fixed it

closed 2019-07-30 by Schild0r