prusa3d/PrusaSlicer #2682: Prusaslicer close when I try big scale.

Blast357, 2019-07-27:
### Version
_Version of PrusaSlicer used 2.0.0+win64-201905201652

### Operating system type + version
Windows 10 home 1809

### 3D printer brand / version + firmware version (if known)
prusa mk3s... but im not getting there :)

### Behavior
I can slice the stl at 100%, 200% no problem... even 300%. but as soon as I try 380% (max for my bed, hum in reality... 382% should be my max) the slicing start and the program just close at the end. no error message, no freezing, just closing when it should be switching to the save/upload button.

I've been able to slice this stl before but I started playing with support and infill a lot. and now... it just close as describe when at 380% even without support...

haven't tried to reinstall yet.

STL USED:
https://www.thingiverse.com/thing:2874382

bubnikv, 2019-07-31:
What is the amount of RAM your machine is equipped with? What is your graphics card? Is it an integrated Intel?

The Gyroid infill generates a huge amount of short moves, which are expensive to visualize. On our Windows computer, the G-code preview takes around 3GB of RAM, the 3D model and the slicing results around 1.5GB RAM, and the time estimator around 1.5GB RAM for the normal mode and another 1.5GB RAM for the silent mode.

You can certainly slice your model if you switch from Gyroid infill to a grid infill. You can lower memory requirements by disabling the "silent mode" time settings in the Printer profile. 

We shall lower memory consumption of the time estimator, but I am afraid it is quite difficult to lower the memory requirements of the 3D print path visualization. There are ways, but only compatible with OpenGL 4.0 and newer.

Blast357, 2019-07-31:
> What is the amount of RAM your machine is equipped with? What is your graphics card? Is it an integrated Intel?
>
> The Gyroid infill generates a huge amount of short moves, which are expensive to [...]

its an old Dell laptop inspiron 17R-SE-7720, i7 3610QM, 16GB of ram, it has an intel HD 4000 and Nvidia 650M video (shared memory).

i've rebooted and tried slicing again and it worked! I tried without support and work fine (ram raised to ~ 4Gb at most)

then I tried with gyroid infill at 20% + support!... the ram jumped to 9.5GB-10GB for a while, and prusaslicer stopped as It did before.

I get a preview in yellow but without the usual color (top infill, overhang, etc...) and it seems to hang there for couples of minutes then the program stop.

would be nice if prusaslicer would tell me something like "Dude, time for a new computer, your ram is full" before closing, but it hardly get to 10GB on 16GB

now even without support, its not working, I may have to try rebooting.

do you really think its a memory issues ? I just tried no support but still 380%... the ram jump to 6.5GB/16GB... I get the yellow preview, seem to hang there and couples of minutes later BOOM it closes on me.

sorry for the long text, trying to describe what is happening

thanks! 

Jose

bubnikv, 2019-08-01:
>  "Dude, time for a new computer, your ram is full" before closing,

We should do that, but handling low memory issues is notoriously difficult.

>  but it hardly get to 10GB on 16GB

I don't know. We certainly found one spot with your example project which
we can improve -> the time estimator memory consumption.

bubnikv, 2021-09-23:
> would be nice if prusaslicer would tell me something like "Dude, time for a new computer, your ram is full" before closing, but it hardly get to 10GB on 16GB

You may have had multiple different applications running. Also Windows expands swap when needed, but it may not be fast enough to do that.

I don't think we can do more for you.
Closing.

closed 2021-09-23 by bubnikv