prusa3d/PrusaSlicer #2683: Layer labeling in GUI and Gcode is very confusing

ke9g, 2019-07-28:
### Version
_Version of PrusaSlicer 1.41.3+

### Operating system type + version
MacOS Mojave 10.14.5

### 3D printer brand not important

### Behavior
FIRST:  The first layer in the GCODE file is called layer zero (0).  WHY?  I am a programmer and understand the concept of arrays starting at index zero but it makes no sense for a real world object like a printer layer to start at zero.  This is very confusing.

SECOND:  In the Plater Preview window, the bottom most visible layer is LAYER TWO (2).  You can't set the slider below the number two.  Why again??  Shouldn't it be layer one (1)?  Or at least to be consistent with your gcode file shouldn't it be layer zero (0)?  This is terribly confusing when trying to set the layer number for a M600 filament change.

THIRD:  The Plater Layers window does not tell you which layer you are looking at.  Yes it gives you the Z height, but that is much less useful than the layer number if you are looking for a specific layer to change color.

_Is this a new feature request?_  NO

#### Project File (.3MF) where problem occurs.  None to submit, just open a Gcode file of your own making.

lukasmatena, 2019-07-28:
I suggest you upgrade to the latest version PrusaSlicer (https://github.com/prusa3d/PrusaSlicer/releases) before filing an issue. This will solve your second issue and also make the M600 insertion easier. It will also kind of solve the third issue, because the layer view was removed.

As for the first issue, I think you should be more explicit. I don't remember there were ever some layer indices in the gcode.

ke9g, 2019-07-28:
Thank you.  I did not know there was "PrusaSlicer".  I was using the latest version of Sli3r Prusa Edition.  This has solved my issue.

closed 2019-07-28 by ke9g