prusa3d/PrusaSlicer #2684: File sliced incorrectly

markdesouza, 2019-07-28:
### Version
PrusaSlicer 2.0.0+

### Operating system type + version
MacOS 10.14.6

### 3D printer brand / version + firmware version (if known)
Prusa MK3s

### Behavior
File slices incorrectly. The attached STL is render correctly in the 3d view (ie the button is solid) but when the file is sliced, the button top is not sliced correctly (images attached)

[image]
[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[wifibutton.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3438929/wifibutton.3mf.zip)
[wifibutton.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/3438930/wifibutton.stl.zip)

lukasmatena, 2019-07-28:
The wall that you're missing is too thin and PrusaSlicer ignores it by default. Enable 'Detect thin walls' in Print Settings->Layers and Perimeters.

markdesouza, 2019-07-28:
Thanks that worked. Is there a disadvantage in having this option enabled?

bubnikv, 2019-07-29:
>  Is there a disadvantage in having this option enabled?

Sometimes slicer will append thin lines to the exterior of the outer shell,
which are not fused well to it. We plan to make the outer thin lines
optional.

jwillmer, 2019-07-29:
@markdesouza can be closed?

markdesouza, 2019-07-29:
sure... and thank you

closed 2019-07-29 by markdesouza