prusa3d/PrusaSlicer #2687: More info: Travel moves & Black Blobs!!

BirkBinnard, 2019-07-28:
### Version
2.0.0

### Operating system type + version
WIn10/64 Home

### 3D printer brand / version + firmware version (if known)
Atom2

### Behavior
1. Excessive travel moves during print
2. Black blobs at the ends of travel moves

_Is this a new feature request?_
No

The topic of excess travel moves is not new, but I'm posting these images to add mor especifics and raise a new but related one. Here is my latest print:
[image]
It printed fine in 42 hours.

But here are 2 images of the inside:
[image]
[image]
The first one shows fine stringing across the interior, and many black blobs at the start (or maybe the end) of the strings. The print settings for this print specified "Avoid crossing perimeters" = Off and "Seam Position Nearest." 

To try to get a better understanding of what was actually happening made this screenshot showing travel moves:
[image]
Next I changed seam position to "Aligned":
[image]
Next I switched to "Avoid crossing perimeters = On" and kept seam position = "Aligned"
[image]
Finally I specified "Avoid crossing perimeters = On" and set seam position = "Nearest"
[image]
Basically I was trying to see if there was anything I could do to eliminate all the crossings since the blobs appeared at the start/end of the crossings. But nothing I tried worked.  So I am not clear on what's really happening here. 
For this print my filament was PLA and my print temperature was 160 C. I don't see how this temperature could get any filament hot enough to turn black. It is true that the walls of this print are fairly thin, but they do have a thickness and I did not use Vase Mode to make the print. (The outer surface is rippled but the inner one is smooth.)

Trellian, 2019-08-14:
There is a real possibility that your temperature sensor is lying to you (maybe not in proper contact with the heatblock), and your hot-end is a lot hotter than you think. The other possibility is that you are using a contaminated filament, and the contaminants are burning.

Both would result in black blobs.

BirkBinnard, 2019-08-14:
Thanks Adrian - I sort of think you are correct. I've been beta-testing a new hotend design and it does use a different approach for the thermistor placement. Unfortunately I don't have any device for accurately measuring my  nozzle temperature, so I'm going to do some experiments and just see how low I can go and still print.

closed 2019-08-14 by BirkBinnard