prusa3d/PrusaSlicer #2688: Unable to compile Master branch with in with Visual Studio 2013 CE

antimix, 2019-07-28:
### Version
Latest Master branch
version_2.0.0-500-gf589e5fb5

### Operating system Windows 10 x64

### Behavior
I am unable to compile the sources.
Error during the build:
"...\PrusaSlicer\build\ALL_BUILD.vcxproj" (destinazione predefinita) (1) ->
"...\PrusaSlicer\build\src\PrusaSlicer.vcxproj" (destinazione predefinita) (3) ->
"...\PrusaSlicer\build\src\libslic3r\libslic3r.vcxproj" (destinazione predefinita) (16) ->
 (destinazione: ClCompile) ->
    ...\prusaslicer\src\libslic3r\pchheader.hpp(105): fatal error C1083: **Cannot open include file: 'cereal/access.hpp'**
       : No such file or directory [...\PrusaSlicer\build\src\libslic3r\libslic3r.vcxproj]

It seems to use the **"cereal"** lib that is not among the sources.

_Is this a new feature request?_ 
NO

#### Project File (.3MF) where problem occurs
none

lukasmatena, 2019-07-29:
Cereal library has been recently added as a dependency. I suppose that updating your dependencies as described in [our wiki](https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Windows.md) should solve it. You'll probably have to compile the dependencies packages yourself. As far as I know, the prebuilt packages for MS Visual Studio 2013 have not been rebuilt yet and since we have upgraded our Windows build system to VS2019, I doubt we will provide the updated version for VS2013. Instead, we should probably remove it. @vojtechkral will correct me if I'm wrong.

vojtechkral, 2019-07-29:
Yes, the pre-built dependencies are outdated and in fact the whole _How to buil on Windows_ doc is outdated as it still recommends MSVS 2013. We'll fix this...

bubnikv, 2019-07-29:
I will provide the prebuilt Win64 dependencies. I think there is no need to
provide Win32 dependencies anymore as only 0.7% of our install base run
Win32 builds.

antimix, 2019-07-29:
- This mean that now I can compile the Master branch with Visual Studio 2019 Community Edition ?
- Do you have the instruction to self-compile the dependences ?

vojtechkral, 2019-07-29:
@antimix 

> This mean that now I can compile the Master branch with Visual Studio 2019 Community Edition ?

Yes.

> Do you have the instruction to self-compile the dependences ?

Well, yes, in that the [How to build on Windows document](https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Windows.md) has a paragraph about compiling the dependencies yourself. It's still applicable, except that you'll need to replace the `"Visual Studio 12 Win64"` generator with one applicable for new MSVS, which would probably be `"Visual Studio 16"` (_without_ the `Win64` suffix).

tamasmeszaros, 2019-07-30:
I'm not sure if the code itself would compile in VS2013 anymore as it lacks full C++11 support.

vojtechkral, 2019-07-30:
Ok, so these two things should probably be done:

- [ ] Add CMake targets to generate the whole deps bundle in the
- [ ] Update the doc

I'll add those onto my to-do stack, but please be patient as I have other higher priority tasks to do at the moment. (Leaving this issue open to track these tasks.)

userosos, 2019-07-30:
I make the self-compile the dependences and it not work for me. Hallo from #2696 bug.

tsamisa, 2019-07-31:
i keep getting this error
CMake Error at E:/Apps/Microsoft Visual Studio/2019/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.14/Modules/ExternalProject.cmake:2427 (message):
  error: could not find git for clone of dep_wxwidgets

antimix, 2019-08-01:
I was able to setup Visual Studio Community Edition 2019, compile the dependencies and finally compile PrusaSlicer with no error.
Since I am relatively new in GitHub usage, what branch / tag should I select to ensure I am compiling the up current newer version ?

lukasmatena, 2019-08-01:
@antimix If you want to follow our development in between releases, follow master branch. But don't expect it to work all the time ;-)

@tsamisa CMake complains that it could not find git. Install git.

tsamisa, 2019-08-02:
@lukasmatena Thanks for the reply. Git was installed. It was a matter of setting windows paths. 

I'm now almost there but now for some reason while building the solution i get that there is no get_io_service identifier (boost libraries).

userosos, 2019-08-02:
> 
> 
> I created a couple of batch file, and I was able to setup Visual Studio Community Edition 2019, compile the dependencies and finally compile PrusaSlicer with no error.
> Since I am relatively new in GitHub usage, what branch / tag should I select to ensure I am compiling the up current newer version ?

Can you make detailed manual, please?

lukasmatena, 2019-08-02:
@tsamisa Yeah, that is quite unfortunate, it was already reported in https://github.com/prusa3d/PrusaSlicer/issues/2150. @vojtechkral commented the code out (it was not currently used), now it should build fine.

antimix, 2019-08-03:
> Can you make detailed manual, please?

Ok, this is not a manual, is just a quick and dirty explanation. 
I build and compile all via Windows Command line, not via IDE, so this is the reason I use batch files.
You need to created the following files, all in the main **PrusaSlicer** folder.

A file named **ENV.BAT** 

`SET PATH=%PATH%;D:\Program Files (x86)\CMake\bin;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Current\Bin
`

then, another file to build the dependencies name **BuildLib64.bat**

```
cd deps
md build
cd build
cmake .. -G "Visual Studio 16" -DDESTDIR="E:\DEVELOP\libWin64\usr\local"
cd ..\..
```

then another to compile the dependencies, named **compLib64.bat**
```
rem
cd deps
cd build
msbuild /m /P:Configuration=Release ALL_BUILD.vcxproj 
cd ..\..
```

another to initialise the PrusaSlicer compilation environment named **INIT.BAT**

```
md build
cd build
cmake .. -G "Visual Studio 16" -DCMAKE_PREFIX_PATH="E:\DEVELOP\libWin64\usr\local\usr\local" -DPERL_LIBRARY="D:\Strawberry\c\lib" -DBOOST_INCLUDEDIR="E:\DEVELOP\libWin64\usr\local\usr\local\include\boost-1_70\boost"
```

and a last one to compile PrusaSlicer, named **comp.bat**
```
cd build
msbuild /m /P:Configuration=Release ALL_BUILD.vcxproj
```

The sequence to create the dependencies is more or less this (you do only once):

- I open the command prompt, 
- I go in the PrusaSlicer folder, and start **ENV.BAT**
- I start **BuildLib64**
- I start **complib64**

then it is time to prepare the PrusaSlicer folders for compilation (just one time):
I start **init**

then your environment is ready. 

You can change the PrusaSlicer sources or just compile them with:
**comp**

I assumed that you have:

- Installed Visual Studio Community Edition 2019
- Installed GitHub
- Installed TortoiseGit
- Installed Boost (http://www.boost.org/)

I hope this will help you. I know that setting up an opensource development environment is one of the most complex things due to the myriad of tools and utilities that "are assumed to be present" in a development pc.

userosos, 2019-08-03:
> Ok, this is not a manual, is just a quick and dirty explanation.
> I build and compile all via Windows Command line, not via IDE, so this is the reason I use batch files.
> You need to created the [...]

Very good! Thx! I can see: "-DPERL_LIBRARY="D:\Strawberry\c\lib" what is this? http://strawberryperl.com/ ?
I installed boost use the manual: https://docs.microsoft.com/en-us/cpp/build/vcpkg?view=vs-2019
vcpkg install boost.

userosos, 2019-08-03:
I has the error still: Building without Win10 Netfabb STL fixing service support
CMake Error at C:/Program Files/CMake/share/cmake-3.15/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find Boost (missing: system filesystem thread log locale regex)
  (found version "1.63.0")
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.15/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  C:/Program Files/CMake/share/cmake-3.15/Modules/FindBoost.cmake:2142 (find_package_handle_standard_args)
  CMakeLists.txt:237 (find_package)

When i run: cmake .. -G "Visual Studio 16" -DCMAKE_PREFIX_PATH="C:\local\destdir-64\usr\local" -DPERL_LIBRARY="D:\Strawberry\c\lib" -DBOOST_INCLUDEDIR="C:\local\destdir-64\usr\local\include\boost_1_70_0\boost"
In "C:\local\destdir-64\usr\local\include\boost_1_70_0\boost" i placed libs from http://www.boost.org/.

userosos, 2019-08-03:
@antimix can you share Assembled slicer? I can not make it into windows it is very crazy for me.

vojtechkral, 2019-08-03:
Downloading Boost manually should not be needed, Boost 1.70 is built as part of the dependencies. Perl shouldn't be needed either. Vcpkg is unlikely to help - we don't use vcpkg.

@userosos Did the dependencies built succesfuly for you? Also, why do you need to build PrusaSlicer from git anyway? There are currently multiple known bugs in the master.

userosos, 2019-08-03:
@vojtechkral I want test it.
C:\Users\user\source\repos\PrusaSlicer-version_2.1.0-alpha0\deps\build>cmake .. -G "Visual Studio 16" -DDESTDIR="C:\local\destdir-custom"
-- Selecting Windows SDK version 10.0.18362.0 to target Windows 10.0.17763.
-- PrusaSlicer deps DESTDIR: C:/local/destdir-custom
-- PrusaSlicer deps debug build: ON
--
Detected 64-bit compiler => building 64-bit deps bundle

-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/user/source/repos/PrusaSlicer-version_2.1.0-alpha0/deps/build

C:\Users\user\source\repos\PrusaSlicer-version_2.1.0-alpha0\deps\build>msbuild /m ALL_BUILD.vcxproj
Build completed successfully.
     Warnings: 0
     Errors: 0

userosos, 2019-08-05:
I have an idea what if make image with ready to build system for virtual machin? It will be easy for build.

bubnikv, 2019-08-05:
>  I have an idea what if make image with ready to build system for virtual
machin? It will be easy for build.

Who will pay for the Windows license?

vojtechkral, 2019-08-05:
Distributed a full-blow VM containing Windows is probably illegal (but IANAL). Regarding lighter-weight containers, I have zero experience with those on Windows. But if you know how to do it (legally), then go for it...

antimix, 2019-08-05:
@userosos I forgoto to mention, that after building the BOOST utility, you should copy the compiled b2.exe into the CMake bin folder (in my case D:\Program Files (x86)\CMake\bin) otherwise the build process will not work.

antimix, 2019-08-05:
@userosos the boost library from http://www.boost.org/ should be used only to compile and build the **b2.exe** program and nothing else. 
The boost library used by PrusaSlicer is self contained on the sources package you get from the prusa repository when you clone PrusaSlicer.

When I use _-DDESTDIR="E:\DEVELOP\libWin64\usr\local"_   "E:\DEVELOP\libWin64\usr\local" is an empty folder just created. The compilation process (**compLib64.bat**) will put there the compiled dependencies.

antimix, 2019-08-10:
Now we use Visual Studio 2019, and I explained above how to do it. I close the issue.

closed 2019-08-10 by antimix

rongith, 2020-03-03:
@antimix 
I wuold like to play with PrusaSlicer develepment too but I'm stuck with the dependencies.
This is my batch file:
```
chcp 437
rem path to cmake.exe
SET PATH=%PATH%;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin
rem path to git.exe
SET PATH=%PATH%;C:\Users\johndoo\AppData\Local\GitHubDesktop\app-2.3.1\resources\app\git\mingw64\bin
rem path to msbuild.exe
SET PATH=%PATH%;C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Current\Bin

cd C:\Users\johndoo\Documents\GitHub\prusa3D\PrusaSlicer\deps\
md build
cd build
cmake .. -G "Visual Studio 16" -DCMAKE_PREFIX_PATH="D:\TempTrav\Dev\libWin64\usr\local"
msbuild /m /P:Configuration=Release ALL_BUILD.vcxproj
```
But I've got errors like that after the msbuild command:
```
         HEAD is now at cacf7f1 zlib 1.2.11
         fatal: 'submodule' appears to be a git command, but we were not
         able to execute it. Maybe git-submodule is broken?
         CMake Error at C:/Users/johndoo/Documents/GitHub/prusa3D/PrusaSlicer/deps/build/dep_ZLIB-prefix/tmp/dep_ZLIB-gitclone.cmake:49 (message):
           Failed to update submodules in:
           'C:/Users/johndoo/Documents/GitHub/prusa3D/PrusaSlicer/deps/build/dep_ZLIB-prefix/src/dep_ZLIB'
```
or
```
       "C:\Users\johndoo\Documents\GitHub\prusa3D\PrusaSlicer\deps\build\ALL_BUILD.vcxproj" (default target) (1) ->
       "C:\Users\johndoo\Documents\GitHub\prusa3D\PrusaSlicer\deps\build\deps.vcxproj" (default target) (3) ->
       "C:\Users\johndoo\Documents\GitHub\prusa3D\PrusaSlicer\deps\build\dep_ZLIB.vcxproj" (default target) (6) ->
       (CustomBuild target) ->
         C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(231,5): error MSB6006: "cmd.exe" exited with code 1. [C:\Users\johndoo\Documents
       \GitHub\prusa3D\PrusaSlicer\deps\build\dep_ZLIB.vcxproj]

       "C:\Users\johndoo\Documents\GitHub\prusa3D\PrusaSlicer\deps\build\ALL_BUILD.vcxproj" (default target) (1) ->
       "C:\Users\johndoo\Documents\GitHub\prusa3D\PrusaSlicer\deps\build\deps.vcxproj" (default target) (3) ->
       "C:\Users\johndoo\Documents\GitHub\prusa3D\PrusaSlicer\deps\build\dep_wxwidgets.vcxproj" (default target) (14) ->
         C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(231,5): error MSB6006: "cmd.exe" exited with code 1. [C:\Users\johndoo\Documents
       \GitHub\prusa3D\PrusaSlicer\deps\build\dep_wxwidgets.vcxproj]

       "C:\Users\johndoo\Documents\GitHub\prusa3D\PrusaSlicer\deps\build\ALL_BUILD.vcxproj" (default target) (1) ->
       "C:\Users\johndoo\Documents\GitHub\prusa3D\PrusaSlicer\deps\build\deps.vcxproj" (default target) (3) ->
       "C:\Users\johndoo\Documents\GitHub\prusa3D\PrusaSlicer\deps\build\dep_CGAL.vcxproj" (default target) (4) ->
         C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(231,5): error MSB6006: "cmd.exe" exited with code 1. [C:\Users\johndoo\Documents
       \GitHub\prusa3D\PrusaSlicer\deps\build\dep_CGAL.vcxproj]
```

I'm kind new to that tools. I managed to compile code when there where still perl codes in Slic3rPE with both cmake and nmake but not now.
I didn't install TortoiseGit nor Boost as it seems unnecessary. I installed github desktop.
> Downloading Boost manually should not be needed, Boost 1.70 is built as part of the dependencies. Perl shouldn't be needed either. Vcpkg is unlikely to help - we don't use vcpkg.

Do you have any recommendation?

bubnikv, 2020-03-04:
Maybe updating cmake will help?