prusa3d/PrusaSlicer #2689: Can't Connect to octoprint via hostname octopi.local

gusdleon, 2019-07-29:
### Version
2.0.0+win64

### Operating system type + version
Windows 10 Pro 64 bits v1903 Comp 18362.239

### 3D printer brand / version + firmware version (if known)
N/A but...
Original Prusa i3 MK3s MMU2s Stock
FW MK3s: 3.7.2
FW MMU2s: 1.0.5

### Behavior
When trying to connect a Octoprint instance via the PrusaSlicer Printer Settings page with the local mDNS hostname (octopi.local) the connection id successful.
[image]

but when trying to upload the gcode, PrusaSlicer fails to upload the file.
[image]

Thanks :)

vojtechkral, 2019-07-29:
I believe this is the same issue as #1781 . Please see therein - you likely need to install Bonjour resolver for Windows. As of now the Bonjour address resolving is not implemented within Slicer.

closed 2019-07-29 by vojtechkral

gusdleon, 2019-07-29:
Hey @vojtechkral Thanks.
I thing i resolved the problem in some manner, I just fount if you disable the feature of UPnP in the router and reboot the router works just fine

**With UPnP enabled**
[image]

**Without UPnP**
[image]

**And here is the .gcode in my Octoprint**
[image]

So I think is not problem of my computer not resolving the hostname, or I´m incorrect? con you explain please

Thanks again :)

vojtechkral, 2019-07-30:
Honestly, I have no idea as to what the relationship between UPnP and mDNS is.

gusdleon, 2019-07-30:
HAHAHAHAHAHA honestly me neither ... but now it works

@vojtechkral
Thanks again.... 🖒

jventerprises, 2020-09-10:
In windows, a hostname that ends with a '.' should force a DNS resolve.  This worked fine for me pre 2.2.0.  This means that my host name of 'octopi.' would resolve in PrusaSlicer.  It still resolves everywhere else.  Now with 2.2.0, its hit or miss in PS.

I just put my IP address instead of the hostname for now

bubnikv, 2021-08-11:
Would you please test this build?
https://github.com/prusa3d/PrusaSlicer/issues/1781#issuecomment-895811943

OctoPrint shall respond to both Bonjour (Apple) and NETBIOS (Microsoft), thus one should not need to install Bonjour on Windows. However there was an issue when sending a G-code from PrusaSlicer, where two HTTP requests were sent to OctoPrint in succession, while NETBIOS was resolved for the first request only, not for the second. We don't know why the NETBIOS resolve fails for the 2nd HTTP request, however we worked around this issue by getting the resolved IP address from the 1st HTTP request and passing it to the 2nd HTTP request, which solved the issue at least in our tests.