prusa3d/PrusaSlicer #2690: [FEATURE REQUEST] Show calculated print time for current layer

jwillmer, 2019-07-29:
Display calculated print time for current layer. This is interesting if I try to bridge some gaps and like to know how long it will take till the bride layer is started. Both to be ready and look how the printer performs at this step as well as to know if it is worth to risk the print in the first place (time wasted).

[image]

JangusKhan, 2019-07-29:
This looks very similar if not functionally the same as request #1621 which is apparently scheduled to be included in an upcoming release.

jwillmer, 2019-07-29:
Closing this in favor of #1621

closed 2019-07-29 by jwillmer