prusa3d/PrusaSlicer #2691: [FEATURE REQUEST] Expose Gap Fill settings for experts

RossKor, 2019-07-29:
### Version
2.0.0+win64

### Requested Features
1. Add expert setting to adjust minimum width for gap fills.
2. Add expert setting to disable gap fills independent of infill (to both support using gap fill with zero infill and to suppress gap fill when using non-zero infill).

### Justification
I'd like to have control over the minimum width for gap-fills. In some cases I've seen "unnecessary" gap fills being employed (e.g. gap fills which I do not feel are needed for the part given its intended application).

Search though I may I did not find any option to control gap fills - other than turning off all infills in which case gap fills also disable.

These "excessive" fills result in increased print times, generally increased stringing, and visibly unappealing for translucent prints.

Jebtrix, 2019-08-02:
I have to give +1 to this request. Gap fill is the one behavior I have to post process out tediously on more intricate parts and certain filaments where it becomes a visible liability in finish quality. 

Sometimes you can get away with just ensuring gap fill is set to a unique speed, say 149.9mm/s. Then use ie _sed_ to strip out gap fill routine with **G1 F8994** (mm/min) as the marker (in verbose output). Scenarios where I wanted to choose gap fill more precisely on a part I had to use modifiers with a unique gap fill speed that I could then strip post process. The omission of gap fill tweak options has always seemed odd to me. PrusaSlicer needs the inverse of "Solid infill threshold area" setting for gap fill and on/off toggle.

Verbose gcode comments should be also be changed to explicit **gap fill** instead of **infill** to help with post processing ambiguity.

ra100, 2019-08-04:
This feature would be most welcome. For some of my prints, gap fill feels more like torture test, and of course unnecessarily extends the print time. I know you can overcome this, as @Jebtrix mentioned, but it would be nice to have option to effortlessly disable or tweak gap fill behaviour.
And I think this is related/duplicate to this: https://github.com/prusa3d/PrusaSlicer/issues/1806

Jebtrix, 2019-08-12:
Well I just realized looking at issue [2746](https://github.com/prusa3d/PrusaSlicer/issues/2746) that there is no need to do post-processing anymore to deal with this. I just never even thought to look at speed. Changing Print Settings > Speed > Gap Fill to **0** will disable gap fill. Works perfectly with modifiers too.

closed 2026-09-29 by prusaslicer-bot[bot]