prusa3d/PrusaSlicer #2692: Feature Request-Adaptive layer thickness only in perimeter (shell)

eid0, 2019-07-29:
Hi, I want to know if anyone has thought in implementing a new way of printing that i have never seen that consists in printing the shell perimeters in smaller thickness continously until reach the normal thickness layer, then make the infill thickness layer normal in one pass. (it can be the other way around).
This way the shell of the piece would be of great quality because it has much more layers than infill and the speed of printing would be very fast as the infill could be thick.
Am I missing anything? Is this possible to implement?
I think it could be revolutionary.
Thanks.

om2kw, 2019-07-31:
did you tried Combine infill every: layers in infill settings?
This might be what you are looking for.

SvetoBobr, 2019-10-20:
Combine infill is completely different concept

rtyr, 2021-02-04:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/402. Closing this one.

closed 2021-02-04 by rtyr